dronekit/dronekit-python #924: Dronekit Not Installing with pip on Raspberry Pi Zero W

Disco1234, 2019-04-05:
I had just completed a full re-install of Raspbian and tried installing dronekit first thing with `pip install dronekit`

I got a mountain of errors including "Could not find function xmlCheckVersion in library libxml2.  Is libxml2 installed?" as well as "Failed building wheel for lxml"

Anyone seen this before?

AdamHtooLwin, 2019-04-17:
There's a few dependences you'll have to install using ```apt``` and I believe this includes libxml.

gabrielruoff, 2019-05-07:
Try
`sudo pip2 install lxml cython jupyter future mavproxy`
before running
`sudo pip2 install dronekit`
You _should_ also be able to run `sudo apt-get install python2-dronekit && sudo apt-get install -f` to fix the missing dependencies but don't me on that until I've checked it on my machine.

maciek01, 2019-10-09:
> I had just completed a full re-install of Raspbian and tried installing dronekit first thing with `pip install dronekit`
>
> I got a mountain of errors including "Could not find function [...]

the same issue on latest stretch pi zero w.
An older install on another pi, performed late 2018 works fine. looks like some braking change got in in between.

Madnyax, 2019-11-11:
$ sudo apt-get install python-pip python-dev
$ sudo apt install libxml2-dev libxslt1-dev (libxslt-dev ではなく)
$ pip install dronekit