dronekit/dronekit-python #925: Moving without GPS

MarsLabAtSLU, 2019-04-09:
I'd like to be able to move and position the drone without GPS, I'm working on a visual positioning system that uses the camera(optiTrack) to detect a target from which position can be determined.

Based on the position read from drone markers(optics),  GPS is being replaced by motion capture coordinates using below API:
 vehicle.message_factory.att_pos_mocap_encode(
            calendar.timegm(gmtime()),
            q,
            x,
            y,
            -1*z); // -1 for increment from ground.

And the I try positioning using below API
self._vehicle.message_factory.set_position_target_local_ned_encode(
            calendar.timegm(gmtime()),
            0, 0,
            1,
            0b0000111111111000,
            x, y, -1*z,
            0, 0, 0,
            0, 0, 0,
            0, 0)

where I am able to arm the drone. But not able to takeoff from ground into air. I did try making sure to increment z axis correctly but still it does not take off after arming it.

Is there any thing that I might be missing for this scenario to work, please do let us know.

MarsLabAtSLU, 2019-04-11:
Any one here to help me on this query of mine ?

skyfloyd, 2020-11-26:
any news here?

luke5799, 2022-03-28:
Well?

proficnc, 2022-03-28:
You need to set a home position

angela804, 2022-07-12:
Have you solved this?