dronekit/dronekit-python #927: Problem with site

lukaszungier, 2019-04-16:
Is the official site down? I cannot access the documentary of this module.

guilhermelac001, 2019-04-16:
The site is down!! Is there any place I can get the documentation?

lukaszungier, 2019-04-16:
I need it ASAP too

victormleon, 2019-04-16:
Using Wayback Machine, the latest crawl is from December 3, 2018. Depending what you’re needing it might have it there. 

https://web.archive.org/web/20181203064336/http://python.dronekit.io/

ShockvaWe, 2019-04-20:
Why is the site down? Is that project desserted?

proficnc, 2019-04-20:
No it is not deserted, it is going through some hosting modification of sorts. 

Lots of people on this.

lukaszungier, 2019-04-20:
Do you know how long it will take to restore the site?

mrthomasbarnard, 2019-04-23:
@proficnc , sorry to bother you, do you know any more about the hosting mods, it would be good to know who was doing them and if they could provide an update to us, it would help a lot, 

thanks