dronekit/dronekit-python #928: dronekit not supporing python3 it is giving error

nagendra092918, 2019-04-18:
please help me it is giving error when i run the any example program.it is giving No module named 'dronekit' . but i test python3 and import dronekit it imported well.
please check the image also exact what is going on.

### **This is the problem i am getting:**
pi@raspberrypi:~/Desktop/dronekit-python/examples/simple_goto $ sudo python3 simple_goto.py
Traceback (most recent call last):
  File "simple_goto.py", line 16, in <module>
    import dronekit
ImportError: No module named 'dronekit'
pi@raspberrypi:~/Desktop/dronekit-python/examples/simple_goto $ python3         Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dronekit
>>>

please help me and tell what is the problem i did any mistake.please please help me

[image]

IuliuNovac, 2019-04-18:
Well, have you installed it under pip3 or pip?

nagendra092918, 2019-04-19:
both i installed pip3 and pip

AdamHtooLwin, 2019-04-21:
I think the issue is with the ```sudo``` when you run the script. Try not using ```sudo```.

If I'm correct, ```pip freeze``` and ```sudo pip freeze``` will output different results. This is because you installed the packages (dronekit included) to your user and not the superuser.

By the way, Python3 isn't supported for dronekit so...
https://web.archive.org/web/20180723144117/http://python.dronekit.io/develop/installation.html

nagendra092918, 2019-04-24:
tq for ur reply. dronekit working on python3 also

closed 2019-04-24 by nagendra092918