dronekit/dronekit-python #932: dronekit(http://python.dronekit.io) website not working. any one have any idea about this.please help

nagendra092918, 2019-04-29:
http://python.dronekit.io
website is not working.please tell any other way to open this website. Thank you
[image]

philtomson, 2019-04-29:
Haven't been able to get to it for at least the last couple of weeks. Also wondering what happened.

hamishwillee, 2019-04-29:
@CraigElder @peterbarker Do you know anything about the (new) hosting of the docs?

In the meantime, you can find instructions for building the docs locally here: https://github.com/dronekit/dronekit-python/blob/master/docs/contributing/contributions_documentation.rst#building-the-docs

gabrielruoff, 2019-05-06:
I've also had trouble with this. You can grab the pages from google's site cache: [](https://webcache.googleusercontent.com/search?q=cache:<your url without "http://">)

Note: it only seems to load if you click "text only".

hamishwillee, 2019-05-06:
Also the wayback machine has a copy.

jakehewitt, 2019-05-08:
Building the docs is failing for me 
https://gist.github.com/jakehewitt/ef8998e6a5c4934c0adeb9ba55001390

jakehewitt, 2019-05-08:
I got it working with a manual build without vagrant. Mirrored it here https://dronekit.netlify.com/

hamishwillee, 2019-05-09:
Thank @jakehewitt - very helpful. FYI, I have spawned off a separate issue for vagrant not working: #937. I do not have time to work out why yet, but I have verified that is is indeed a problem.

mwshafer, 2019-05-09:
Is there any timeline for when the documentation is going to be back online?

hamishwillee, 2019-05-09:
> Is there any timeline for when the documentation is going to be back online?

@CraigElder is AFAIK looking into this.

mwshafer, 2019-05-10:
Thanks for the reply. Just to clarify, is @CraigElder looking into when the documentation will be back online, is he working on getting it back online? Thanks.

CraigElder, 2019-05-20:
@mrpollo Hi Ramon
Do you have an update on your documentation?

mrpollo, 2019-05-21:
Here's a summary, the website is now hosted on github [dronekit/dronekit.io](https://github.com/dronekit/dronekit.io/), sadly I lost access to my GoDaddy account where I hosted the DNS, I'm working it out with their support, as soon as I regain control we can have a website again, the documentation is hosted on the source code of each project, we just need to make sure we have a continuous build system preferably running on the current CI infrastructure, contributions greatly appreciated, although we won't be able to have proper sub domains until the DNS is resolved.

I have been posting updates on Gitter.IM for anyone interested in following progress.