dronekit/dronekit-python #933: Dronekit timeout after 30 seconds while connecting

developermastermind07, 2019-05-02:
Connecting to vehicle on: 127.0.0.1:14551
Traceback (most recent call last):
  File "arm.py", line 34, in <module>
    vehicle = connect(connection_string, wait_ready=True)
  File "C:\Program Files\python\lib\site-packages\dronekit\__init__.py", line 31
72, in connect
    timeout=timeout)
  File "C:\Program Files\python\lib\site-packages\dronekit\__init__.py", line 23
75, in wait_ready
    timeout)
dronekit.TimeoutError: wait_ready experienced a timeout after 30 seconds.

Using mavproxy i connected to UAV/drone which is connected COM^6 port. Using out option i am forwarding output to 127.0.0.1:14551 port. When dronekit trying to connect to this one throwing timeout exception

gabrielruoff, 2019-05-03:
I was having this issue earlier and fixed it by removing all versions of dronekit and mavproxy and then making sure I reinstalled them using pip2: `pip2 install mavproxy dronekit dronekit-sitl`. When you use `pip` it defaults to installing the python3 version (if you have python3, whic I assume you do) and then when you run your code with `python` it also defaults to python3. After reinstalling, I made sure I ran all dronekit code with `python2` ex: `python2 arm.pi --connect 127.0.0.1:14551` Also, I noticed you're using `MISSION` as a vehicle mode. It should be `GUIDED` :) Finally, I prefer to use sim_vehicle.py to run SITL, and then just connecting to the udp out port when running your python code.
**Running sim_vehicle.py**
As your user
`cd ~`
`git clone https://github.com/ArduPilot/ardupilot.git`
`ardupilot/Tools/autotest/sim_vehicle.py -v ArduCopter`

**Connecting to SITL run from sim_vehicle.py**
`python2 arm.pi --connect udp:127.0.0.1:14550`
or in python:
`connection_string = "udp:127.0.0.1:14550"`

I find using sim_vehicle.py to be much more stable than running SITL and mavproxy by themselves.
Hope this helped.
-Gabe

developermastermind07, 2019-05-03:
@gabrielruoff : Yes i am using python3. Same script works for dronekit-sitl but not working actual UAV.
Also when i tried to connect directly to USB port using above script still it fails to connect. ex python2 arm.pi --connect /dev/ttyUSB0

gabrielruoff, 2019-05-03:
>Yes i am using python3. Same script works for dronekit-sitl but not working actual UAV.
Also when i tried to connect directly to USB port using above script still it fails to connect.

If you're connecting to an actual vehicle you need to pass the port to which the vehicle is connected through --connect, not '127.0.0.1' which is a local network address. Only SITL runs on 127.0.0.1, not the actual vehicle. If you're using USB try `python2 arm.py --connect /dev/ttyUSB0` or in python: connection_string = '/dev/ttyUSB0'

Guohao91, 2019-12-18:
@developermastermind07 I have a similar problem, shown like:
"Traceback (most recent call last):
  File "drone_delivery.py", line 232, in <module>
    vehicle = connect(connection_string, wait_ready=True)
  File "/home/clickmox/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 3172, in connect
    timeout=timeout)
  File "/home/clickmox/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 2375, in wait_ready
    timeout)
dronekit.TimeoutError: wait_ready experienced a timeout after 30 seconds."
And I intend to change the  connection_string = '/dev/ttyACM0' in python script.also tried with python2 ....py. Doesn't work at all so far.
I heard itis because I should upgrade pymavlink or sth like that. I have no idea now.
Looking forward help!

gnsharans, 2021-11-18:
Try keeping wait ready False
It worked for me

mioho-wby, 2023-03-12:
> 

it's a great suggestion ,Thanks!