dronekit/dronekit-python #935: Force Mavlink Version

custeve, 2019-05-06:
In pymavlink, I do the following and it helps force the version to MAVLINK 2.0: 

```
os.environ['MAVLINK20'] = '1'
os.environ['MAVLINK_DIALECT'] = 'ardupilotmega'
```
However, the same trick does not seem to have an effect in DroneKit. Is there a simple way to force MAVLINK 2.0 instead of 1.0 on default? I am running a custom modfiied version of DroneKit already, so changes to the dronekit package are acceptable in my case. We have a unique comms setup, that boots up before the aircraft. It seems that in some cases, if DroneKit boots first, it will default and stay on MAVLINK 1.0.

custeve, 2019-05-07:
I think I may have hacked this into working by adding the following line after the mav connection is verified: 

```
self.vehicle._master.first_byte = True
```

wynn4, 2021-02-24:
@custeve 

Can you verify that this hack is still your standby for forcing dronekit to use Mavlink 2.0? For this to work, do you need to make changes to pymavlink?

It seems like there should be a way to select Mavlink 2.0 in dronekit and I'm surprised there isn't an issue about this.

Amith4504, 2021-04-14:
Any updates on how to set mavlink protocol version to 2.0 using dronekit ?

GRS26, 2023-04-06:
did anybody solve this?