dronekit/dronekit-python #936: Drifting off in Loiter Mode

gabrielruoff, 2019-05-06:
I'm running dronekit on a raspberry pi connected to a pixhawk 4.
Yesterday I was testing some code that makes the drone take off, move to a local position, and loiter for 30 seconds while detecting objects below it using a camera and tensorflow. The vehicle took off correctly and moved to the local position, but then when it switched to loiter mode it continued to drift away until it hit a tree branch and shut off the motors completely (although power to the vehicle was not lost). It appeared as if it didn't switch to loiter at all but just continued to drift in the same manner that it had when flying to the relative position. This was very confusing as I had previously run my code in SITL with flightgear to visualize the drone's movements and it behaved exactly as expected. Additionally, when it began to drift off, I triggered the ctrl-C failsafe I'd written into my python code, which did nothing. In previous tests (simulated and with the real vehicle) the failsafe has worked every time. My theory is that the Raspberry Pi lost connection to the Pixhawk, however, I did not get the typical 'no heartbeat' message in the terminal running my python code. It's worth noting that I often get some drift in loiter mode, but I attribute it to the lack of satellite reception at my house (I usually get ~9 sats) and the vehicle will correct itself after drifting a few feet. Also, the drift I observed here was faster than the usual drift. I'd also calibrated my sensors before running this mission.

**Here's the python2 code I'm running that handles all the heavy lifting (Tensoflow communicates with the python2 code over a socket on localhost:12000 to send it the required data):**
`#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
© Copyright 2015-2016, 3D Robotics.
simple_goto.py: GUIDED mode "simple goto" example (Copter Only)

Demonstrates how to arm and takeoff in Copter and how to navigate to points using Vehicle.simple_goto.

Full documentation is provided at http://python.dronekit.io/examples/simple_goto.html
"""

#Set up option parsing to get socket options
import argparse
import socket
import errno
from object_detection import tcpclientsimple
from dronekit import connect, VehicleMode, LocationGlobalRelative
from pymavlink import mavutil
import time
import signal
import sys
import sys
import subprocess
import getpass

parser = argparse.ArgumentParser(description='Print out vehicle state information')
parser.add_argument('--socket',
                   help="use socket?")
parser.add_argument('--connect',
                    help="vehicle connection target string")
args = parser.parse_args()

useSocket = args.socket

def safeReceive(msglen=3):
    recv = ''
    try:
        recv = server_socket.receive(msglen)
    except socket.error as serr:
        print('encountered error')
    return recv

def receiveClassName():
    server_socket = tcpclientsimple.socketClient()
    server_socket.connect('', 12000)
    recvlen = int(safeReceive())
    print('recvlen: ', recvlen)
    time.sleep(0.25)
    server_socket = tcpclientsimple.socketClient()
    server_socket.connect('', 12000)
    return safeReceive(recvlen)

#connect to socket and wait for init
if useSocket:
    print('using socket')
    server_socket = tcpclientsimple.socketClient()
    server_socket.connect('', 12000)
    print('connected to socket. waiting for init (0x1)....')
    recv = safeReceive()
    while recv != '0x1':
        server_socket = tcpclientsimple.socketClient()
        server_socket.connect('', 12000)
        time.sleep(0.2)
        if safeReceive() == '0x1':
            break
[... 493 lines cut ...]
        if GetClassName(data) is not None:
            sendClassNameOverSocket(GetClassName(data))
        else:
            sendClassNameOverSocket('None')

        cv2.putText(frame,"FPS: {0:.2f}".format(frame_rate_calc),(30,50),font,1,(255,255,0),2,cv2.LINE_AA)

        # All the results have been drawn on the frame, so it's time to display it.
        cv2.imshow('Object detector', frame)

        t2 = cv2.getTickCount()
        time1 = (t2-t1)/freq
        frame_rate_calc = 1/time1

        # Press 'q' to quit
        if cv2.waitKey(1) == ord('q'):
            break

        rawCapture.truncate(0)

    camera.close()

cv2.destroyAllWindows()`

Any help with this is much appreciated!

hamishwillee, 2019-05-06:
> Edit: Sorry for the bad code formatting I'm not really sure how to get it to display correctly..

Three backticks - view source change above.

What I would do is put in some telemetry output to confirm that you really are in loiter mode when you think you are, and possibly to output health info including GPS position.
If you are in loiter mode and you're drifting, and you can verify that, it becomes a question that you can raise on the ArduPilot discussion board.

gabrielruoff, 2019-05-06:
> Three backticks - view source change above.

Thanks! I'll remember that in the future.

> What I would do is put in some telemetry output to confirm that you really are in loiter mode when you think you are, and possibly to output health info including GPS position.
> If you are in loiter [...]

Alright. The tlog was corrupted during the crash, but I have the output of the goto_and_record_objects.py which returns `print('Mode: ', str(vehicle.mode.name))`. Is that enough to assume that the vehicle was actually in loiter?

hamishwillee, 2019-05-06:
> Is that enough to assume that the vehicle was actually in loiter?

The current mode is cached. However if you have a connection then if you're still getting the same mode in 5 seconds or so I'd imagine it is correct.

Basically you want to get as much telemetry as you can to help the flight stack team debug, so confirming mode and everything else you can about health will help.

gabrielruoff, 2019-05-07:
> The current mode is cached. However if you have a connection then if you're still getting the same mode in 5 seconds or so I'd imagine it is correct.
>
> Basically you want to get as much telemetry as [...]

Thanks for the advice. I'll rebuild the quad and try to reproduce the issue with a simple takeoff -> loiter -> land mission as well as write the python's debugging output to a logfile. I'll also video it for good measure so I only have to try and reproduce this once hahaha.

kaangoksal, 2019-05-08:
I got a crash in simple loiter as well, It was due to high hdop and gps glitches. If you are flying close to surfaces that can reflect gps signals (buildings) and have bad gps reception, loiter can result into incosistent behavior from time to time. I had flights that were fine for 5 minutes and then turned crazy around 10. Also you should have a remote controller emergency as well (switch to stabilize)