dronekit/dronekit-python #940: Ardupilot+Dronekit+Gazebo

guilhermelac001, 2019-05-14:
I'm trying to simulate the simple_go_to code on the Iris model with ardupilot plugin for gazebo. I wanted to know if there is any support for dronekit because every time I try to connect with the drone, I get the error: 

[Err] [Connection.hh:266] Header is empty

hamishwillee, 2019-05-14:
@guilhermelac001 There is no properly resourced support or ongoing maintenance, but there is quite an active community that do spot fixes and help with questions. The best way to talk to that community is often https://gitter.im/dronekit/dronekit-python

For this case, it is a new error on me. You might want to better outline your setup/configuration - ie version of ArduPilot, version of DroneKit python etc. Where is this error being reported - DroneKit?

cakircays, 2019-08-19:
Hi. I am trying to write a code that releasing a payload in defined coordinate for drone. I will use raspberry pi and can you help me about the code. I can not find what i need in dorenkit library.