dronekit/dronekit-python #945: dronekit.APIException: mode (33816576, 157) not available on mavlink definition

dwiel, 2019-05-31:
The exception `dronekit.APIException: mode (33816576, 157) not available on mavlink definition` occurs when connecting to a PX4 simulation after calling `self.vehicle.simple_takeoff(altitude)` during the `HEARTBEAT` message handler. It appears that `mavutil.interpret_px4_mode` is returning the mode `TAKEOFF`, but `self._mode_mapping` only contains `['MANUAL', 'STABILIZED', 'ACRO', 'RATTITUDE', 'ALTCTL', 'POSCTL', 'LOITER', 'MISSION', 'RTL', 'LAND', 'RTGS', 'FOLLOWME', 'OFFBOARD']`.

library versions:
`dronekit (2.9.2)`
`pymavlink (2.3.4)`

peterbarker, 2019-06-02:
On Fri, 31 May 2019, Zach Dwiel wrote:

> The exception dronekit.APIException: mode (33816576, 157) not available on mavlink definition occurs when connecting to a PX4 simulation after calling
> self.vehicle.simple_takeoff(altitude) during [...]

Huibean Lo added support for TAKEOFF: 
https://github.com/ArduPilot/MAVProxy/pull/614

This probably isn't in the most recent pymavlink release...

It's in the 2.3.5 pymavlink release...

dwiel, 2019-06-02:
Very strange, thanks for letting me know. Not sure why I had an old
version, I only built this docker image in the past couple of weeks.

Thanks again and sorry to bother you with this

closed 2019-06-02 by dwiel