dronekit/dronekit-python #946: dronekit.connect timeout > 30 not effective

dwiel, 2019-06-05:
I would like to specify a timeout > 30 seconds, but if I do, I get the following exception after 30 seconds anyway:

```
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/dronekit/mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "/usr/local/lib/python3.6/dist-packages/dronekit/__init__.py", line 1371, in listener
    self._heartbeat_error)
dronekit.APIException: No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "/tmp/wait_for_simulator_then_quit.py", line 11, in <module>
    main()
  File "/tmp/wait_for_simulator_then_quit.py", line 7, in main
    connect("udpin:localhost:14540", wait_ready=True, timeout=300)
  File "/usr/local/lib/python3.6/dist-packages/dronekit/__init__.py", line 3166, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/usr/local/lib/python3.6/dist-packages/dronekit/__init__.py", line 2275, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
```

this occurs after 30 seconds, even though i specified a timeout of 300 seconds.

amalpavithran, 2019-06-14:
This should stop the Exception from being thrown
`vehicle = connect("/dev/ttyUSB0", wait_ready=False, baud = 57600)`
`vehicle.wait_ready(True, raise_exception=False)`
Not sure if this is how it's to be used but have found this to work in our case

dwiel, 2019-06-14:
Cool thanks I'll try that. Right now I just have my own retry loop around the connection function, which works alright as well. This looks cleaner though.

Guohao91, 2020-01-03:
@dwiel Hi ,Sir. I think I am sticking with the same problem as you did. I already tried what @amalpavithran said, but still:
[image]
So have you improved anything a bit? Looking forward your reply! So appreciate it!
I am so annoyed and tired of investigating these problems.
I am using Pixhawk 4 btw.

amalpavithran, 2020-01-03:
@Guohao91 Are u using telemetry or USB. Since we were having some issues on telemetry we had actually shifted to using an onboard raspberry pi connected to the telemetry port. Try using a USB cable and confirm whether this issue still occurs.

Guohao91, 2020-01-03:
@amalpavithran  I tell you what. I keep trying on both usb port on the side and TELEM2. The problem i posted here is with TELEM2 status.
And if i turn to usb port condition, it shows up like:
[image]
it will keep going forever....
My compnay's onboard NUC hasn't come over, so i connected my desktop(Linux Ubuntu) instead to the flight controller.

Guohao91, 2020-01-03:
@amalpavithran well so far, I say USB port works better than TELEM2. Because MAVProxy works almost fine with USB port, while TELEM2 doesn't detect out my UAV at all.

amalpavithran, 2020-01-03:
@Guohao91 I think the wp index out of bounds issue is the same as #725. I think it is similar to mavros requesting home position but that shouldn't interfere with the functionality (Do correct me if I am wrong). I feel that an onboard computer and communicating over a faster channel like websockets is a better setup (kinda working on this not tested yet).

Guohao91, 2020-01-03:
yes, i saw that posted issue. Apparently, @hamishwillee  is saying the issue comes up when we running any dronekit script, which is also what i am doing. We can't even successfully to run a script, I've been bothered this for the past 3 months.

IIAB-prog, 2020-01-04:
@amalpavithran

      Great your code allowed me to solve the same problem.raspberry connect to pix4 via telemetry.
Thanks!!

anwar1awad, 2020-01-05:
can anyone please help me out , am using an emax f3 FC , connected to the uart of the bananapi (similar to raspberry pi zero ) and i couldnt figure it out how to get connected and data communucation between the 2 boards , i want to use the telemtry port , @amalpavithran  , can you please share how could you work it out ? 
thanks alot guys

amalpavithran, 2020-01-05:
@anwar1awad Sorry I haven't worked with the above board but I think the issue here is that the Emax F3 FC does not support the mavlink protocol on which dronekit API is based upon. Here is some docs on the protocols supported https://github.com/betaflight/betaflight/blob/master/docs/Telemetry.md

anwar1awad, 2020-01-05:
@amalpavithran thanks for your rapid response ,
for the FC , i flashed it to betaflight , so it supports mavlink now , and connected to betaflight config. and it works

thanapong-skru, 2022-04-25:
> This should stop the Exception from being thrown `vehicle = connect("/dev/ttyUSB0", wait_ready=False, baud = 57600)` `vehicle.wait_ready(True, raise_exception=False)` Not sure if this is how it's to be used but have found this to work in our case

You are  the great. I can solve a problem, no heartbeat in last 5 second as your recommendation, thank you so much, please tell me your full name, I have to cite in my report.

Varun-UTS, 2024-07-31:
Hi @amalpavithran 
I have been working on something similar and getting the same error. So I am using rfd900 on both sides. I tried injecting a MAVLink packet using the Python script but got a link timeout. Also, I was using Mission Planner on the other side. I even tried it with baud = 57600 and timeout = 30, got same result. If you have any idea please let me know.
Below is the script I used

from dronekit import connect, VehicleMode, mavutil
import time

def connect_to_vehicle(connection_string):
    """ Connect to the vehicle using the provided connection string """
    print(f"Connecting to vehicle on {connection_string}...")
    vehicle = connect(connection_string, wait_ready=True, baud=912600,timeout=120)  # Adjust the baud rate if necessary
    return vehicle

def send_global_position(vehicle):
    """ Send a MAVLink GLOBAL_POSITION_INT message """
    # Create a MAVLink message
    msg = vehicle.message_factory.global_position_int_encode(
        int(time.time() * 1000),  # time_boot_ms (milliseconds since boot)
        374709983,  # lat (latitude in 1E-7 degrees)
        -1224194153,  # lon (longitude in 1E-7 degrees)
        150000,  # alt (altitude in mm, above MSL)
        150000,  # relative_alt (altitude above ground in mm)
        0,  # vx (ground X speed in cm/s)
        0,  # vy (ground Y speed in cm/s)
        0,  # vz (ground Z speed in cm/s)
        0  # hdg (heading in 1E-2 degrees)
    )

    # Send the MAVLink message
    print("Sending GLOBAL_POSITION_INT message...")
    vehicle.send_mavlink(msg)
    vehicle.flush()

def main():
    connection_string = 'COM3'  # Replace with your actual COM port
    vehicle = connect_to_vehicle(connection_string)

    try:
        while True:
            send_global_position(vehicle)
            time.sleep(1)  # Send the message every second

    except KeyboardInterrupt:
        print("Stopping script")

    finally:
        vehicle.close()
        print("Connection closed")

if __name__ == "__main__":
    main()