dronekit/dronekit-python #950: No data with dronekit - but get data with python serial multiwii

diyquad, 2019-06-28:
Hi,
I'm trying to get dronekit working on a PI Zero + Omnibus F4 with Inav.
For now i'm connected with USB.

When i launch dronekit, i got errors
`device reports readiness to read but returned no data `

But i know communication beetween FC and PI is working, since i'm also trying with the pyMultiwii module, and python "classic" multiwii  and i have data, and can send data to RC.

Seams only not working when doing it with dronekit .

Also, strange thing: each time I launch and break the process (because of errors) i also noticed that after the keyboard interrupt, the /dev/ttyACM0 is not more mounted (ls /dev/tty*: no show the ACM0 after keyboard interrupt, but was showing before i launch dronekit)

If you have any idea

Thanks a lof for your help.

**CODE**
`
from dronekit import connect, VehicleMode
connection_string='/dev/ttyACM0'
print("Connecting to vehicle on: %s" % (connection_string,))
vehicle = connect(connection_string, wait_ready=True)

try:
    print "Get some vehicle attribute values:"
    print " Last Heartbeat: %s" % vehicle.last_heartbeat
    print " Is Armable?: %s" % vehicle.is_armable
    print " System status: %s" % vehicle.system_status.state
    print " Mode: %s" % vehicle.mode.name    # settable

except KeyboardInterrupt:
    print "Cancelled"
    vehicle.close()

vehicle.close()
`

**And console return:**
`WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:dronekit.mavlink:Exception while receiving message:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 218, in mavlink_thread_in
    msg = self.master.recv_msg()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 420, in recv_msg
    s = self.recv(n)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 962, in recv
    ret = self.port.read(n)
  File "/home/pi/.local/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
`

gamer-12748, 2021-04-10:
the device probably doesn't communicate in the format required by dronekit.

gamer-12748, 2021-04-10:
i think you can change the format in either inav cli, or raspberry pi python

gamer-12748, 2021-04-10:
also it has been a year so check if it works now by any chance?