dronekit/dronekit-python #951: New install of dronekit fails during import / /ardupilotmega.xml

GregoryVPerry, 2019-06-29:
Stock Ubuntu 18.04 LTS install, after pip3 install DroneKit and pip install DroneKit, both python2 and python3 fail with the following error:

IOError: [Errno 2] No such file or directory: 'message_definitions/v1.0/ardupilotmega.xml'

peterbarker, 2019-06-30:
On Sat, 29 Jun 2019, GregoryVPerry wrote:

> Stock Ubuntu 18.04 LTS install, after pip3 install DroneKit and pip install DroneKit, both python2 and python3 fail with the following error:
> 
> IOError: [Errno 2] No such file or directory: 'message_definitions/v1.0/ardupilotmega.xml'

Can you post the full output of the command, please?

GregoryVPerry, 2019-06-30:
```
$ python
Python 2.7.15+ (default, Nov 27 2018, 23:36:35) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from dronekit import connect
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/gperry/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 45, in <module>
    from pymavlink import mavutil, mavwp
  File "pymavlink/mavutil.py", line 110, in <module>
    set_dialect(os.environ['MAVLINK_DIALECT'])
  File "pymavlink/mavutil.py", line 101, in set_dialect
    mavgen_python_dialect(dialect, wire_protocol)
  File "pymavlink/generator/mavgen.py", line 244, in mavgen_python_dialect
    if not mavgen(opts, [xml]):
  File "pymavlink/generator/mavgen.py", line 136, in mavgen
    if not mavgen_validate(fname):
  File "pymavlink/generator/mavgen.py", line 105, in mavgen_validate
    with open(xmlfile, 'r') as f:
IOError: [Errno 2] No such file or directory: 'message_definitions/v1.0/ardupilotmega.xml'
>>> 
```

peterbarker, 2019-06-30:
On Sat, 29 Jun 2019, GregoryVPerry wrote:

> >>> from dronekit import connect

Unsurprisingly, that works for me.

Might I suggest repeatedly running "pip uninstall pymavlink", "pip3 
uninstall pymavlink", "pip uninstall dronekit", "pip3 uninstall dronekit" 
and they all report no package by that name is installed - and *then* 
install them?

I've got a suspicion there might be older files floating around...