dronekit/dronekit-python #959: Handler error

goldern, 2019-07-23:
After updating dronekit from version dronekit-2.9.1 to dronekit-2.9.2 and pymavlink-2.0.6 to pymavlink-2.3.7, my python script is outputting the following error after trying to connect to openpilot revolution on com7:
```
No handlers could be found for logger "dronekit"
```
and with SITL:
```
Start simulator (SITL)
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
 No handlers could be found for logger "autopilot"
```
How do I define 'handlers'(whatever that is, I have no clue) in order to make that error stop?

peterbarker, 2019-07-24:
@pietrodn @Huibean do either of you have an idea on this one?

@goldern Do the examples work appropriately?

goldern, 2019-07-25:
@peterbarker yeah, they do work, but it's weird that after an update that handler error would keep appearing, even if there's a handler set.

Huibean, 2019-07-29:
@peterbarker sorry for late reply, I can not get this error reappear yet with both python2 and python3

pietrodn, 2019-08-25:
Try to add:

```python
import logging
logging.basicConfig()
```

on top of your code. This should create a default handler for the root logger.

closed 2021-02-17 by goldern