dronekit/dronekit-python #960: SITL shows no reaction from RC override

srediske, 2019-07-24:
Hey,

i'm working with sim_vehicle.py and Gazebo and want to implement an easy RC_override in mode STABILIZE but it seems that i do something wrong.

`time.sleep(3)
print("Set Ch3 override to 1900")
vehicle.channels.overrides[3] = 1900
print(" Channel overrides: %s" % vehicle.channels.overrides)
time.sleep(10)`

The Drone has no reaction to this code, its in stabilize mode and armed. I can override with mavproxy (rc x yyyy) but i want to make it work with dronekit or mavlink only. The MAV command 
`msg = vehicle.message_factory.command_long_encode(0, 0, mavutil.mavlink.MAV_CMD_DO_SET_SERVO, 0, 2, 1900, 0, 0, 0, 0, 0) 
vehicle.send_mavlink(msg)`
shows at least a reaction from SITL: "Channel 2 is already in use"
Its only armed and in stabilized, nothing more.

Any Ideas?

peterbarker, 2019-07-24:
On Wed, 24 Jul 2019, srediske wrote:

> i'm working with sim_vehicle.py and Gazebo and want to implement an easy
> RC_override in mode STABILIZE but it seems that i do something wrong.
>
> time.sleep(3) print("Set Ch3 override to 1900") [...]

Does the channel_overrides example work?

srediske, 2019-07-25:
The example doesn't do anything except printing in the terminal. But the values look too low, cause in SITL the minimum RC Values for RC 1,2,3,... are at PWM 1000 and maximum are at 2000. In the example from dronekit are values about 300.

srediske, 2019-07-25:
Hey guys, i found the problem. I already changed the parameter **SYSID_MYGCS** from **255** to **1** for some work with mavproxy rc override and doesn't know that i have to change it back to 255 for rc override with dronekit/mavlink.

May it help a stranger who will one day strumble upon this thread ;-)

closed 2019-07-25 by srediske

angela804, 2022-07-12:
I'm trying to implement this one. My RPi and PixHawk are connected through TELEM 2 and RX TX
I'm looking for a way where Raspberry pi can somehow make the servos move so I hope this works out tomorrow.