dronekit/dronekit-python #965: SITL ignores PITCH changes after set_attitude_target_encode()

artardupilot, 2019-08-01:
Hey guys! 

When calling `set_attitude_target_encode()` for changing ROLL it works correctly (vehicle moves left/right). But it doesn't work for changing PITCH, I mean it completely ignores the message, and so PITCH stays unchanged. 

MODE: GUIDED
VEHICLE_TYPE: PLANE

Changing ROLL:
`q = quaternion.QuaternionBase([np.radians(45), np.radians(0), np.radians(0)])`
`msg = vehicle.message_factory.set_attitude_target_encode(`
`0,`
`0,`
`0,`
`0b11100010,`
`q,`
`0,`
`0,`
`0,`
`0)`
`vehicle.send_mavlink(msg)`
`vehicle.flush()`

Changing PITCH:
`q = quaternion.QuaternionBase([np.radians(0), np.radians(45), np.radians(0)])`
`msg = vehicle.message_factory.set_attitude_target_encode(`
`0,`
`0,`
`0,`
`0b11100010,`
`q,`
`0,`
`0,`
`0,`
`0)`
`vehicle.send_mavlink(msg)`
`vehicle.flush()`

peterbarker, 2019-08-01:
On Thu, 1 Aug 2019, artardupilot wrote:

> When calling set_attitude_target_encode() for changing ROLL it works correctly (vehicle moves left/right). But it doesn't work for changing PITCH, I mean
> it completely ignores the message, and so PITCH stays unchanged.

> 0b11100010,

       <field type="uint8_t" name="type_mask">Mappings: If any of these 
bits are set, the corresponding input should be ignored: bit 1: body roll 
rate, bit 2: body pitch rate, bit 3: body yaw rate. bit 4-bit 6: reserved, 
bit 7: throttle, bit 8: attitude</field>

I think your mask here needs help.

artardupilot, 2019-08-02:
Thanks for your reply.

PITCH and ROLL bits are both set in mask, and if there is something wrong in mask, then ROLL shouldn't work either.

Also there is an interesting behavior when working with CMD instead of dronekit.
I can change PITCH from CMD with mavproxy using `attitude` command.
After executing this command the nose of the plane tilts up/down but after a while the vehicle itself starts to fly down/up changing altitude until it equals to the altitude before the `attitude` command.

Here is example command to tilt the vehicle (plane) up by 45 degrees:
`attitude 0.923 0.0 0.328 0.0 0.0`

closed 2019-08-23 by artardupilot

domnantas, 2021-03-13:
This was a problem with `type_mask`. The `type_mask` sets which parameters are **ignored**. Also, read it from right to left. If you want to enable roll, pitch yaw and throttle, the `type_mask` should look be `0b10111000`. [MAVlink docs](https://mavlink.io/en/messages/common.html#ATTITUDE_TARGET_TYPEMASK)

More info in another issue: https://github.com/dronekit/dronekit-python/issues/1044#issuecomment-798777347

Muhammetyilmaz7, 2024-05-01:
q = quaternion.QuaternionBase([np.radians(0), np.radians(45), np.radians(0)])
print(q)
msg = vehicle.message_factory.set_attitude_target_encode(
0,
0,
0,
0b11100010,
q,
0,
0,
0,
0)
vehicle.send_mavlink(msg)
vehicle.flush() 
What exactly is done in this code and what does quaternion.QuaternionBase do? I am using dronekit and how do I integrate it into my own code?