dronekit/dronekit-python #966: Manual Override

priyankamadhavan, 2019-08-06:
I am working on a project where we are trying to implement communications between 2 drones. The basic structure is: Drone1 sends a command to Drone2 (through UDP and Python Socket) to first arm, then takeoff, and fly to up 5 meters. Drone2 then sends a message to Drone1 to fly up 5 meters. When testing this, the altitude sensors on our drones can be iffy, which results in them flying higher than intended, and after about 30 seconds they end up crashing down. 

We want to create a manual takeover command so we can use our RC Remotes to pilot the drones if something like this were to happen again. It seems like the only way to do this is with channel overrides, which is strongly advised against in the documentation. We are able to run RC commands on the drones, but when running our message code, it seems like the RC is disabled. 

Thank you!!

hamishwillee, 2019-08-06:
Why don't you just change the mode to loiter/hold in the fail state? Then switch to whatever RC mode you like or back to offboard from there.

priyankamadhavan, 2019-08-09:
Yes, I changed it to loiter mode and it worked! It seems like in guided mode the command has to run before receiving another. Thank you!

closed 2019-08-09 by priyankamadhavan