dronekit/dronekit-python #967: About RC Channel 9-16

thorstenbar, 2019-08-08:
Hello,
currently its only possible to read the first 8 RC-Channels that come from the Controller. I was wondering why it is not implemented for the whole 16 channels that mavlink supports? And another thing: Is there an easy way to override these higher channels, because i noticed, that pymavlink currently only implemented 8 channel overriding.

liutairan, 2021-02-26:
Have you gotten a solution for this issue?

KimJinHyeok1, 2022-02-10:
I got a Solution!

Edit the Dronekit Library!

In the Python library folder and Enter the Dronekit folder and Edit __init__.py and modify the ChannelsOverride Class.

Code Snippet
---------------------------------------------------------------------------------------------
class ChannelsOverride(dict):
    """
    A dictionary class for managing Vehicle channel overrides.

    Channels can be read, written, or cleared by index or using a dictionary syntax.
    To clear a value, set it to ``None`` or use ``del`` on the item.

    An object of this type is returned by :py:attr:`Vehicle.channels.overrides <Channels.overrides>`.

    For more information and examples see :ref:`example_channel_overrides`.
    """

    def __init__(self, vehicle):
        self._vehicle = vehicle
        self._count = 10  # Fixed by MAVLink
        self._active = True

    def __getitem__(self, key):
        return dict.__getitem__(self, str(key))

    def __setitem__(self, key, value):
        if not (0 < int(key) <= self._count):
            raise KeyError('Invalid channel index %s' % key)
        if not value:
            try:
                dict.__delitem__(self, str(key))
            except:
                pass
        else:
            dict.__setitem__(self, str(key), value)
        self._send()

    def __delitem__(self, key):
        dict.__delitem__(self, str(key))
        self._send()

    def __len__(self):
        return self._count

    def _send(self):
        if self._active:
            overrides = [0] * 10
            for k, v in self.items():
                overrides[int(k) - 1] = v
            self._vehicle._master.mav.rc_channels_override_send(0, 0, *overrides)
---------------------------------------------------------------------------------------------

In the above code you just only edit 2 line

    self._count = **(The number you Want)**  # Fixed by MAVLink
    overrides = [0] *  **(The number you Want)** 

good luck!

AyushMaria, 2023-09-28:
Any update on this?