dronekit/dronekit-python #969: Getting 'None' for all channel values

phil-toppe, 2019-08-26:
Hey guys!

I followed [this](http://ardupilot.org/dev/docs/sitl-native-on-windows.html) guide to setup SITL and flightgear (for planes). Now I am able to launch the simulator and load my Tarananis X9D joystick with mavproxy command `module load joystick` to control the plane. I am also able to connect to the simulator from the phone using QGroundControl with udp.

Now I want to connect to the simulator from a python script using dronekit-python lib and read all rc channels values. Here is my script:
```python
from dronekit import connect
import cv2

def channels_callback(self, attr_name, value):
	 newChannels = value
	 print("Ch1 from callback: %s" % newChannels['1'])
	 print("Ch2 from callback: %s" % newChannels['2'])
	 print("Ch3 from callback: %s" % newChannels['3'])

# Connect to the Vehicle
print('Connecting to vehicle')
vehicle = connect('udp:0.0.0.0:14555', wait_ready=True, timeout=90)
print('Connected')

# Get all original channel values
print("Channels:", vehicle.channels)

vehicle.add_attribute_listener("channels", channels_callback)

while True:
	print("Ch1: %s" % vehicle.channels['1'])
	print("Ch2: %s" % vehicle.channels['2'])
	print("Ch3: %s" % vehicle.channels['3'])
	if cv2.waitKey(1) & 0xFF == ord('q'):
		break
				
#Close vehicle object before exiting script
print("\nClose vehicle object")
vehicle.close()

print("Completed")
``` 

The connection part is always successful, but the channel values reading fails. In the while loop all 3 `print` statements always print 'None' and the callback function is never called. The channels original values are also 'None':
```python
# Get all original channel values
print("Channels:", vehicle.channels)
```

Any help would be appreciated.

mrthomasbarnard, 2020-01-28:
Sadly I'm having the same problem, even though it is with code that has been working for some time now.  It now report None for all channels

avncalst, 2020-03-22:
I had the same problem and solved it by using a decorator on_message listener:
   ```
   
         @self.vehicle.on_message('RC_CHANNELS')
         def chin_listener(self, name, message):
            # print '%s attribute is: %s' % (name, message)
            global pitch
            global roll
            global mod
            global flag_rec

            
            pitch=message.chan2_raw    # copter (throttle)
            roll=message.chan1_raw     # copter (angle)
            # throttle=message.chan3_raw # copter (throttle) 
            mod=message.chan6_raw      # fly, stop, shift left, right, switch SD
            flag_rec=message.chan7_raw # drive - record mode, switch SG
            # print 'channels:',pitch,roll,mod,flag_rec
```
Remark: it does not work for all messages

avncalst, 2020-03-22:
my vehicle is defined in a python class, therefor I refer to "vehicle" as "self.vehicle".

saminhasan, 2022-05-22:
Can you please share the whole vehicle class

jakobjackson2016, 2022-09-27:
@phil-toppe were you able to find out and solve this issue? I too am getting None values for all my 8 channels. Thanks!

mrthomasbarnard, 2022-09-28:
@jakobjackson2016, I found a work around for the time when the channels return None.  Its a shame that from time to time the channels report none even though the information is correctly being sent and received by drone kit.

In short I use a decorator when the channels return None to load a global which seems to work reliably for me

global PH_Channels
PH_Channels = None

	def Channel(self,chNum):
		if self.vehicleConnected is True:
			if self.vehicle.channels[1] == None :
				try:
					if PH_Channels == None:
						if self.vehicle.channels[7] == None:
							self.VE.Inform("Channel are None - decorator being deployed on RC_Channels",1,"AP")

							
							@self.vehicle.on_message('RC_CHANNELS')
							def RCIN_listener(self, name, message):

								global PH_Channels

								tempchan = []
								tempchan.append(0)
								tempchan.append(message.chan1_raw)
								tempchan.append(message.chan2_raw)
								tempchan.append(message.chan3_raw)
								tempchan.append(message.chan4_raw)
								tempchan.append(message.chan5_raw)
								tempchan.append(message.chan6_raw)
								tempchan.append(message.chan7_raw)
								tempchan.append(message.chan8_raw)
								tempchan.append(message.chan9_raw)
								tempchan.append(message.chan10_raw)
								tempchan.append(message.chan11_raw)
								tempchan.append(message.chan12_raw)
								tempchan.append(message.chan13_raw)
								tempchan.append(message.chan14_raw)

								PH_Channels = tempchan

						channel = None
					else:
						channel = PH_Channels[chNum]
				except:
					channel = None
			else:
				channel = self.vehicle.channels[int(chNum)]
		else:
			channel = None
		return channel

angela804, 2022-10-23:
Are there any causes of this one? I was able to access channels in my R-Pi. When I tried to download another firmware in my Pixhawk, the values return None.

angela804, 2022-10-24:
Can I ask how do you put this one in full code @avncalst 

> I had the same problem and solved it by using a decorator on_message listener:
>
> ```
>
>       @self.vehicle.on_message('RC_CHANNELS')
>       def chin_listener(self, name, message):
>          # print '%s [...]

mrthomasbarnard, 2022-11-25:
It's a know bug, best way round it it to use a decorator to get the values
when None is returned

Tom