dronekit/dronekit-python #973: Raspberry pi Drone

ABDULBASIT1998, 2019-09-08:
Hey,
I am building a autonomous drone and a newbie in the drone community. I got a python code for autonomous takeoff and landing. I also configured my pixhawk with raspberry pi. I just wanted to ask–
Is radio setup in Mission planner/Qgroundcontrol is compulsory for the initial setup???
also—
Without any kind of radio setup I was trying to run the python code, the code works perfectly but there is no any kind of motor rotations???

please help…

leonardocfor, 2019-09-09:
Hey man, you need to do the initial setup e.g. compass calibration, GPS, etc., including radio setup in order to have the PX4 ready. Afterwards, you can try your Python code and everything should be fine.

ABDULBASIT1998, 2019-09-10:
Thank you so much for ur reply.....
Does it works in the same way for the Ardupilot?

leonardocfor, 2019-09-11:
What do you mean? 

PX4 is the hardware but you can run on it PX4 software or ArduPilot and the initial setup is pretty much the same using Qgroundcontrol or Mission planner. You have to take the drone and spin around haha, etc.

ABDULBASIT1998, 2019-09-11:
Sorry I meant it as pX 4 software..Thank u so much...

leonardocfor, 2019-09-11:
No pb man :)