dronekit/dronekit-python #977: Error trying to change modes

acwatkins, 2019-10-15:
I am running a mavgateway that connects over serial to a pixhawk 2. The companion computer is connecting over udp with the connection string "udpout:127.0.0.1:14550". This code has been working for a while before today. I get the following error when I run:

```
vehicle.mode = dronekit.VehicleMode("AUTO")
```

```
>           self._master.set_mode(self._mode_mapping[v.name])
E           TypeError: 'NoneType' object has no attribute '__getitem__'

/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py:1621: TypeError
------------------------------ Captured log setup ------------------------------
ERROR    dronekit:__init__.py:1533 Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
APIException: mode (0, 64) not available on mavlink definition
ERROR    dronekit:__init__.py:1533 Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
APIException: mode (0, 64) not available on mavlink definition
```

I refreshed my docker image and used pymavlink-2.4.0. I tried downgrading to 2.3.8 with no change.

I messed with different GCSs connected during initialization and no GCSs connected, but got the error. It rarely doesn't happen, so it seems to  be somewhat intermittent.

acwatkins, 2019-10-16:
vehicle._vehicle_type is sometimes 0 (when it throws an exception) and 14 when it works.

acwatkins, 2019-10-16:
I believe I figured out the issue. A mavlink camera was connected that was sending out a vehicle type of 0. This was occasionally setting vehicle_type to 0, which caused an exception when changing modes.

wynn4, 2021-05-05:
Hi @acwatkins I am occasionally seeing this same problem. Would you mind sharing your solution? Were you able to keep your mavlink camera connected and still overcome this issue?

KarlEaves, 2021-05-26:
@acwatkins Were you able to resolve this? We are having the same exact error show up on v2.9.2 when an Entire Air Commander is attached

acwatkins, 2021-05-27:
@wynn4 @KarlEaves I simply disconnected the data cable from the camera, as I didn't need a mavlink connection to it.