dronekit/dronekit-python #980: Error when I running takeoff.py

alexgermanotta, 2019-10-20:
hello
i have a problem when i running takeoff program.
i get the next message:

 python takeoff.py
Connecting to vehicle on: 192.168.1.72:14550
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1371, in listener
    self._heartbeat_error)
APIException: No heartbeat in 180 seconds, aborting.
Traceback (most recent call last):
  File "takeoff.py", line 12, in <module>
    vehicle = connect(args.connect, baud=921600, wait_ready=True,heartbeat_timeout=180)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 3166, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2275, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

I use raspberry pi 3b+ and pixhawk
I change timeout 30 second to timeout 180 second, but i can`t fix it.

I can also do that pixhawk connect to mission planer throw UDP.

i followed this guide : http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html 

thanks

leonardocfor, 2019-10-20:
Hi Alex,

How are you starting mavproxy? You can do it like this:

mavproxy.py --master=/dev/serial0 --baudrate 921600 --out=udp:127.0.0.1:14551 (provided that you have followed the required configuration in the ardupilot.org link)

With this approach, you are mapping the serial port communication to your localhost at the port 14551. And then you don't need to specify baud rate in your connect line, i.e.

vehicle = connect('127.0.0.1:14551', wait_ready=True,heartbeat_timeout=180)

It has worked for me using Raspberry PI 3 model B running Ubuntu 18.04 and Pixhawk running Ardupilot

alexgermanotta, 2019-10-20:
Hi leonardo, thanks for you answer.

I deleted all my raspberry pi ,and installed all libraries again.

 I got the same message, perhaps I`m confused whit the settings. I have the next code:

`import argparse  
parser = argparse.ArgumentParser()
parser.add_argument('--connect', default='192.168.1.72:14551')
args = parser.parse_args()`

` #Connect to the Vehicle
print 'Connecting to vehicle on: %s' % args.connect
 vehicle = connect('192.168.1.72:14551', wait_ready=True,heartbeat_timeout=180)`

in this lines i put the IP address of my raspberry

I starting mavproxy as yout told me

` mavproxy.py --master=/dev/serial0 --baudrate 921600 --out=udp:192.168.1.74:14551`

int this line, I used the IP address of my pc to send data to mission planer through UDP and get the parameters of pixwauk

Sorry if my questions are very basic, but I'm new to this
Thank you

nagendra092918, 2019-11-02:
enable the serial port in raspberry pi b+, in b+ the serial port used by Bluetooth so, u should disable the bluetooth serial port, then run the code u will get

LordHandLee, 2020-01-30:
I am having the same issue when I try to connect to localhost and run scripts locally. I tried all suggestions above as well as other suggestions found elsewhere and nothing works. I am using latest version of dronekit and arducopter. What could be the root cause of dronekit not being able to establish connection to localhost? Faulty hardware?