dronekit/dronekit-python #985: Excception in message  for Heartbeat each time i connect the Gimal

JorgeCosta87, 2019-11-11:
Hello, I'm having an error each team i connect the gimbal (STorM32), to the pixhawk :

**APIException: mode (6, 0) not available on mavlink definition
ERROR:dronekit:Exception in message handler for HEARTBEAT**

anyone knows how to solve this ?

Guohao91, 2019-11-14:
same problem

bobn3, 2020-01-14:
Hi, I've updated dronekit, pyMavlink, and my pixhawk's FW and I'm still getting those messages when connected with a storm32 gimbal. Anyone got any idea how to fix it?

kiranshila, 2020-01-16:
It sure does seem like this project was abandoned. The last update to master was in November and I submitted the PR to fix this issue back then. Who knows what's happening.

bobn3, 2020-02-14:
Hi, I'm having this same problem. Pixhawk 2.1 with Storm32 gimbal, using latest dronekit and ardupilot versions, followed the steps here and verified that the gimbal does respond to RC's ch7 and moves accordingly, however, when trying it out with python it doesn't move (it does read the gimbal position correctly though), and it raises this exception:

    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (6, 128) not available on mavlink definition