dronekit/dronekit-python #993: Use dronekit in micropython / pycom project

bart, 2019-11-29:
Hey everyone,

I really like the features dronekit provides when talking on a raspberry pi as a companion computer to pixhawk. In a new project Im trying to use a pycom fipy inseatd of a pi which has running micropython on it. So I asked myself: Is it possible to port the dronekit python libs w/ mavlink to micropython? What do you think?

Really looking forward your opinion and ideas, thanks in advance!

ijustwant, 2019-11-29:
Great idea! I would like the same.

RichardBH, 2020-06-27:
Bart, I'm interested aswell. I curently use Pycom modules to do some telemetry. I don't have much experience with mavlink. And am just starting in RC/UAV planes.
I am really interested in setting up the FiPy with communications via LTE(CATM1), with failover to Lorawan and Wifi if available.
At first I was just looking at the pycom as a comms module for telemetry and comms from ground station to dronekit running on a pi zero w. But having Dronekit run in the Pycom would be good for small aircraft but may not have the processing power of the pi.
Have you started the port?

bart, 2020-06-27:
Hey @RichardBH, I didn't start the port but built my own FC instead based on Arduino. It can talk to any custom sensor or I/O device (like the FiPy) via I2C wich for me is the best solution. I know it's way more complicated than using an existing project but porting Dronekit to Micropython for me would be more work than building my own perfect Flight Controller that exactly work like I want him to work. So sorry on this. Think somebody else has to pick up this topi and may build a Micropython version.

RichardBH, 2020-06-27:
Thats fine. I agree with you, the pycom doesnt really have the grunt for dronekit so would make a poor substituse. if I can ask whcih arduino platform are you using and how far have you come with your FC, (flying?) I'm really interested because I am looking to buy an FC. I have an APM from 10 years ago, but hardware has changed a lot since then. I'm kinda swinging between buying something off the shelf or make my own.

bart, 2020-06-28:
Depends on what you want to achieve. Fo a usual drone it makes sense to go with a Pixhawk for example which you can also control by companion computer. I built mine based on an ESP32 board with lots of additional sensors like GPS, openMV image processing, sonar, lidar etc. and a custom 2.4 GHz transceiver for telemetry and manual input like kill switch capability. Everything else is built totally with autonomous flying by design. 3D printed landing legs yesterday so I hope I can do the first takeoff during the next few days. On a scale from 1 to 100 I'm maybe at 50% because now the software fine tuning starts which takes a lot of time.

closed 2020-06-28 by bart