dronekit/dronekit-python #996: Can't connect dronekit scripts to my solo copter(Pixhawk4 controller), although the example scripts work all fine with SITL instance

Guohao91, 2019-12-18:
when I try to run any one example script onto my UAV(Pixhawk4 controller), it doesn't work well, problem shown as:
[image]
All the examples work fine with SITL instance, as well as onto my Mavproxy.
So i am quite confused why my UAV can connect mavproxy while not the example scripts.
and I am pretty sure my port "/dev/ttyACM0" is correct.

closed 2020-01-09 by Guohao91

mohamadsaab, 2020-03-27:
hi , did you find a solution ? I'm facing the same problem

Guohao91, 2020-03-31:
@mohamadsaab hi! sorry for so late reply. 
so i don't remember what reason it was exactly, but i am sure the problem should came out of two possible scenarios:
no.1:in the flight controller side, you need to choose either micro usb port or TELEM2 port, i am using micro usb port to connect this time while I heard some companion computers don't work with TELEM2 port. so when i use TELEM2 port to connect with, flight controller always fails to connect my script.
no.2:   in the NUC(companion computer) side, there are usually 3 or 4 usb ports on the computer right? so I probably made a mistake to connect a port with identifying a wrong port name, like the port i was connecting is ACM1, but i typed : /dev/ttyACM0. so it failed to callback my script.
Hope above explanations can help you even a bit.

mohamadsaab, 2020-03-31:
Ok I understand . I am using navio2 which is HAT for raspberry pi so i can't change the port unfortunately . I thought it would be another reason but thank you for replying.