dronekit/dronekit-python #997: Channel override in landing mode

deepdigger, 2019-12-28:
Hi together, 
in the wiki they say that you should open an issue before trying to use channel override.

Our usecase:
We are a students group developing a drone that should pick objects up. To center on the objects we're setting the vehicle-velocity setting in GUIDED-Mode with dronekit. However, the guided-mode doesn't seem to be suitable for landing, as the motors don't turn off automaticly (we tried it once and the copter flipped and crashed, but that was like 6 months ago). 
That's why we need to use the landing-mode at some point (the copter should center all the way till right before the motors disarm).  However, the landing mode doesn't support the send vehicle-velocity settings, the only option nowadays is to use the channel-override method.

Maybe is there an option we missed to check if the drone has already landed, so that we can use guided-mode? (like in the fc-software). As now I don't see a better way as to use channel-override ([specification for landing mode](https://ardupilot.org/copter/docs/land-mode.html) )

peterbarker, 2019-12-30:
On Sat, 28 Dec 2019, deepdigger wrote:

> We are a students group developing a drone that should pick objects up. To center on the objects we're setting the vehicle-velocity setting in GUIDED-Mode
> with dronekit. However, the guided-mode [...]

Perhaps you could look at using precision landing?

Here's ArduPilot's autotest: 
https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L3729

(additionally, you should probably specify which flightstack you're using; 
PX4Firmware or ArduPilot are typical :-)  )

deepdigger, 2019-12-30:
Hi Peter, thanks for your answer :).
Sorry, I missed saying our flightstack. We're using the Pixhawk cube with Ardupilot. 

I saw the precision landing but thought it would only work with an IR-sensor (from [ardupilot documentation](https://ardupilot.org/copter/docs/precision-landing-with-irlock.html) ).

However, I didn't knew of the other way by sending mavlink messages and specifing the landing target. :)
So thanks for giving us the hint!

deepdigger, 2020-01-03:
I got it to work:). Important thing is to have the height in the message (like specified [here](https://github.com/patrickpoirier51/JeVois--Python-Tracking/blob/master/Precision_Landing_Jevois.py#L30) and to enable this feature in the settings. 
Thanks!

closed 2020-01-03 by deepdigger