dronekit/dronekit-python #999: I would like to make fly my intel ready to fly drone with dronekit, without GPS and without Remote Control. 3D fix ?

clemvoyou, 2020-01-15:
Hello,

I am a beginner on dronekit-python.

I would like to fly indoor. So I would like to make fly my intel ready to fly drone with dronekit, without GPS and without Remote Control. 

But the fonction vehicle.simple_takeoff only work in GUIDED mode. How can I put my drone 
fly at a constant altitude ? Is it possible ? 

In addition, sometimes the drone say me that he need 3D fix. What is it ? 

When the drone will be at the target altitude it will must make a simple fly like:
2 meters in front
2 meters back
Return to launch.

How can I do this ? Is it possible to  acting on the channels ?

This is my code: 

from dronekit import connect, VehicleMode
import time, os

vehicle = connect('tcp:127.0.0.1:5760', wait_ready=False)

def arm_and_takeoff(aTargetAltitude):
 
    print("Arming motors")
	
    vehicle.mode = VehicleMode("GUIDED_NOGPS")
    vehicle.armed = True

    while not vehicle.armed:
        print(" Waiting for arming...")
        time.sleep(1)

    print("Taking off!")
    vehicle.simple_takeoff(aTargetAltitude) 

    
        time.sleep(1)

time.sleep(8)

print("Returning to Launch")
vehicle.mode = VehicleMode("RTL")

# Close vehicle object before exiting script
print("Close vehicle object")
vehicle.close()

Motors speen but thez don t take any power !!!!

Someone can help me ???

Thanks and sorry for my english, i am french !!!!!!!!!!!

Guohao91, 2020-01-23:
@clemvoyou Hi, fellow. We have a similar project scenario ...I need to fly my hexa-multicopter underground(no satellite signals). And I equip a companion computer on my drone, running with dronekit then, no remote control of course.
So:
when drone says he need 3D fix. that basically means your GPS module doesn't get enough signals from satellites (at least 4 required).
And so far my motors just pulse a bit, not working actually. it is not caused by codes, it is because of your wiring or firmware issue.
Overall, dronekit codes can barely cause problems. Pure code work is easy to be fixed. The physical wiring and firmware always have problems, that is the hardest part.

supericeandfire, 2020-10-21:
try mode GUIDED_NOGPS