dronekit/dronekit-python #1000: Cannot connect to real drone running Arducopter via dronekit-python

gaetanokent, 2020-01-16:
Hi Everyone,

I’ve been trying without success to connect with a simple dronekit-python script to my drone running **navio2+rp3** and Arducopter. Connection to the drone works perfectly with both **QGroundControl** or **APMPlanner** … **I connect directly via the Wifi of the Raspberry** since both the drone and my laptop running QGC or APM are connected to the same wifi router at home.

On Ardupilot running on Navio2 I have edited the file sudo nano /etc/default/arducopter with the configuration **#TELEM1="-A udp:192.168.1.108:14550"** where 192.168.1.108 is the IP address of my laptop.

The droneKit script I’m using is the basic:

**vehicle = connect(‘192.168.1.114:14550’, wait_ready=True)** 

where 192.168.1.114 is the IP address of the drone on the wifi network. This script is saved on my Connect.py script and I run it on my laptop (where normally I run QGC or APM)

**Unfortunately this is not working.** I tried both uptions: using tpc and upd (when using tpc I simply use ‘tcp:192.168.1.114:14550’ when calling the connect function). Both are not working. I get:

**Case UDP:**
Traceback (most recent call last):
  File "Connect.py", line 15, in <module>
    vehicle = connect('192.168.1.114:14550', wait_ready=False)
  File "/Library/Python/2.7/site-packages/dronekit/__init__.py", line 3159, in connect
    handler = MAVConnection(ip, baud=baud, source_system=source_system, source_component=source_component, use_native=use_native)
  File "/Library/Python/2.7/site-packages/dronekit/mavlink.py", line 130, in __init__
    self.master = mavutil.mavlink_connection(ip, baud=baud, source_system=source_system, source_component=source_component)
  File "/Library/Python/2.7/site-packages/pymavlink/mavutil.py", line 1670, in mavlink_connection
    return mavudp(device, source_system=source_system, source_component=source_component, input=input, use_native=use_native)
  File "/Library/Python/2.7/site-packages/pymavlink/mavutil.py", line 1015, in __init__
    self.port.bind((a[0], int(a[1])))
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 49] Can't assign requested address

**Case TCP:**

[Errno 61] Connection refused sleeping
[Errno 22] Invalid argument sleeping
Traceback (most recent call last):
  File "Connect.py", line 15, in <module>
    vehicle = connect('tcp:192.168.1.114:14550', wait_ready=False)
  File "/Library/Python/2.7/site-packages/dronekit/__init__.py", line 3159, in connect
    handler = MAVConnection(ip, baud=baud, source_system=source_system, source_component=source_component, use_native=use_native)
  File "/Library/Python/2.7/site-packages/dronekit/mavlink.py", line 130, in __init__
    self.master = mavutil.mavlink_connection(ip, baud=baud, source_system=source_system, source_component=source_component)
  File "/Library/Python/2.7/site-packages/pymavlink/mavutil.py", line 1636, in mavlink_connection
    use_native=use_native)
  File "/Library/Python/2.7/site-packages/pymavlink/mavutil.py", line 1169, in __init__
    self.do_connect()
  File "/Library/Python/2.7/site-packages/pymavlink/mavutil.py", line 1189, in do_connect
    raise e
socket.error: [Errno 22] Invalid argument

The reason why I need to use dronekit-python is to get telemetry data from the drone and upload missions written in code, directly from my laptop. **Do you know where am I doing a mistake? Should I configure something different in Arducopter?**

mohamadsaab, 2020-03-27:
hi , i'm facing the same problem .Did you find a solution ?

agising, 2020-04-02:
When connecting dronekit to sitl the procedure is the following: 
in sitl add output 192.168.1.10:14550. (the ip of the computer you want to connect from)
in dronekit, connect to 192.168.1.10:14550 (yes, your own ip)

you could give it a try (i.e. connect to 192.168.1.108:14550).

I don't understand how my computer finds the way to sitl. If someone can explain I'm happy to listen :)