dronekit/dronekit-python #1005: Upload waypoints weird rounding issue

catokass, 2020-02-20:
When I upload to the Pixhawk via Mission Planner, and then download it again, there doesnt seem to be any issue. But when I use Dronekit to upload a mission file, it comes back looking very strange.

Here's the (simplified) code I use to upload:
`upload_mission_file("D:\TestMission.txt")`

`# Upload mission file
def upload_mission_file(file_name):
    mission_list = load_mission(file_name)

    # Clear existing mission from vehicle
    print(' Clear mission')
    cmds = communications_global.vehicle.commands
    cmds.clear()

    # Add new mission to vehicle
    for command in mission_list:
        cmds.add(command)

    #for command in cmds:
    #    print(str(command))

    print('Upload mission')
    communications_global.vehicle.commands.upload()
    print("Uploaded mission")`

`# Load mission from file
def load_mission(file_name):
    # Load a mission from a file into a list. The mission definition is in the Waypoint file
    # format (http://qgroundcontrol.org/mavlink/waypoint_protocol#waypoint_file_format).
    # This function is used by upload_mission().

    print("\nReading mission from file: %s" % file_name)
    cmds = communications_global.vehicle.commands
    missionlist = []
    with open(file_name) as f:
        for i, line in enumerate(f):
            if i == 0:
                if not line.startswith('QGC WPL 110'):
                    raise Exception('File is not supported WP version')
            else:
                linearray = line.split('\t')
                ln_index = int(linearray[0])
                ln_currentwp = int(linearray[1])
                ln_frame = int(linearray[2])
                ln_command = int(linearray[3])
                ln_param1 = float(linearray[4])
                ln_param2 = float(linearray[5])
                ln_param3 = float(linearray[6])
                ln_param4 = float(linearray[7])
                ln_param5 = float(linearray[8])
                ln_param6 = float(linearray[9])
                ln_param7 = float(linearray[10])
                ln_autocontinue = int(linearray[11].strip())
                cmd = Command(0, 0, 0, ln_frame, ln_command, ln_currentwp, ln_autocontinue, ln_param1, ln_param2,
                              ln_param3, ln_param4, ln_param5, ln_param6, ln_param7)
                missionlist.append(cmd)
    return missionlist`

--

Here's what I upload:
`

catokass, 2020-04-01:
Fixed:

After reading a lot of source code I came upon a very simple solution. 

The [Command](https://github.com/dronekit/dronekit-python/blob/fe6c62313831c7d4847ae8009af3f6400aea8c5e/dronekit/__init__.py#L2871) class in `__init__.py` uses [mavutil.mavlink.MAVLink_mission_item_message](https://mavlink.io/en/messages/common.html#MISSION_ITEM) for the message type, which uses the less accurate float32 for sending the long and lat parameters. This can be made more accurate if we use the [mavutil.mavlink.MAVLink_mission_item_int_message](https://mavlink.io/en/messages/common.html#MISSION_ITEM_INT) type which uses int32 for sending, and is the recommended implementation as per [this](https://mavlink.io/en/services/mission.html#command_message_type). 

We then just have to multiply the global latitude and longitude (degrees) by 10^7 when creating the commands. This can be altered in the [example ](https://github.com/dronekit/dronekit-python/blob/master/examples/mission_import_export/mission_import_export.py)like this:
```
linearray=line.split('\t')
ln_index=int(linearray[0])
ln_currentwp=int(linearray[1])
ln_frame=int(linearray[2])
ln_command=int(linearray[3])
ln_param1=float(linearray[4])
ln_param2=float(linearray[5])
ln_param3=float(linearray[6])
ln_param4=float(linearray[7])
ln_param5=int(float(linearray[8]) * 1e7)
ln_param6=int(float(linearray[9]) * 1e7)
ln_param7=float(linearray[10])
ln_autocontinue=int(linearray[11].strip())
```

If this could be added to the documentation that would be great.

Hope this will help someone in the future.

closed 2020-04-01 by catokass

kimdongchulseo, 2020-06-26:
> Fixed:
>
> After reading a lot of source code I came upon a very simple solution.
>
> The [...]

It seems like a good idea.
After modifying 'mission_import_export.py', what should I do next?

catokass, 2020-06-29:
> It seems like a good idea.
> After modifying 'mission_import_export.py', what should I do next?

Find the installation folder of dronekit and locate `__init__.py`, find the `Command` class shown in [this link](https://github.com/dronekit/dronekit-python/blob/fe6c62313831c7d4847ae8009af3f6400aea8c5e/dronekit/__init__.py#L2871), and replace the argument `mavutil.mavlink.MAVLink_mission_item_message` with `mavutil.mavlink.MAVLink_mission_item_int_message`. This will tell dronekit to use the int type instead of the float type message.