dronekit/dronekit-python #1007: ImportError: cannot import name connect

Jakkev, 2020-02-26:
Hello,

I use a Raspberry Pi 4 with Raspian Buster.
I tried to install Dronekit, but every time I try to run a python program I get the error 
ImportError: cannot import name connect. 
And it does not matter if i write:  from dronekit import connect
or just:  import dronekit
Always the same error.
I checked pip list and its in the list.

I already looked here on Github and Stackoverflow. But unfortunately none of the tips I found helped.
I also tried with different pip's pip2 install, pip3 install, as well as with pip install git+https://github.com/dronekit/dronekit-python.
And to start the script with python and python3.
Installing Dronekit within virtualenv unfortunately did not solve the problem either.

At the end I only want the GPS coordinates, altitude and bearing of the Pixhawk. If I can retrieve these data without dronekit, that would be also enough for me.

I hope you guys can help me somehow.

leonardocfor, 2020-02-26:
Hi Jakkev. Can you share the code in the Python script you are running?

Jakkev, 2020-02-26:
Jeah of course.

#import dronekit
from dronekit import connect, VehicleMode
vehicle = connect('/dev/serial0', wait_ready=True, baud=921600)

It's only to try the import.

leonardocfor, 2020-02-26:
I don't know if it is a typo, but otherwise you should remove the hashtag (#) from the line #import dronekit (the # is commenting the line therefore it is not being executed) and it should work. 

A second thing should be kept in mind regarding the port (/dev/serial0) you are using to connect to the vehicle in order to work. Follow the instructions in this link to do so. 

https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html

Jakkev, 2020-02-26:
Thank you for your help.

I followed the link you send me before and also tried Mavproxy.py and this works.
I have only problems with dronekit.

For the script I typed, I used this example:
https://github.com/dronekit/dronekit-python
And i also tried exactly this script from dronekit, but unfortunately I get the same error.

With "import dronekit" I wanted to try if I can import the whole dronekit library without "from dronekit import connect".
But also there comes the same error:
ImportError: cannot import name 'connect' from 'dronekit

leonardocfor, 2020-02-26:
Can you open the python interpreter and just run  import dronekit. If it does not work, it means dronekit is not installed. That is the pip/pip3 command did not work. In such case, you can try to install it again and paste here the output of the pip/pip3 command.

Jakkev, 2020-02-28:
sorry for the late response.

I found my dumb mistake. I named the file for the test dronekit.py and this caused the error.
After I renamed it worked.

Thank you for your help and time.

simeneide, 2021-11-20:
oh my god, exact same mistake here 🤦

neilpotdukhe23, 2022-01-25:
same mistake here hahaha

heleidsn, 2024-03-11:
same mistake here 🤦