dronekit/dronekit-python #1008: RC Overrides: To visually test each motor

ankurash, 2020-03-06:
I understand that the RC overrides are not recommended. 
However, I want to add a safety check before flight where it can be visually verified that each motor works and sounds normal, before flight. For that I need to override RC and command each motor to run in a sequence at given PWM values.
If it's possible to do that now, any guidance is appreciate.
Otherwise, this is a feature request.

peterbarker, 2020-03-11:
On Fri, 6 Mar 2020, Ankur A Sharma wrote:

> I understand that the RC overrides are not recommended.
> However, I want to add a safety check before flight where it can be visually verified that each motor works and sounds normal, before flight. [...]

Which flight stack?

ArduPilot supports

       <entry value="209" name="MAV_CMD_DO_MOTOR_TEST" hasLocation="false" 
isDestination="false">
         <description>Mission command to perform motor test.</description>
         <param index="1" label="Instance" minValue="1" increment="1">Motor 
instance number. (from 1 to max number of motors on the vehicle)</param>
         <param index="2" label="Throttle Type" 
enum="MOTOR_TEST_THROTTLE_TYPE">Throttle type.</param>
         <param index="3" label="Throttle">Throttle.</param>
         <param index="4" label="Timeout" units="s" 
minValue="0">Timeout.</param>
         <param index="5" label="Motor Count" minValue="0" 
increment="1">Motor count. (number of motors to test to test in sequence, 
waiting for the timeout above between them; 0=1 motor, 1=1 motor, 2=2 
motors...)</param>
         <param index="6" label="Test Order" enum="MOTOR_TEST_ORDER">Motor 
test order.</param>
         <param index="7">Empty</param>
       </entry>

ankurash, 2020-03-11:
Yep, I found it later. Thanks. I'll close the issue now.

closed 2020-03-11 by ankurash