dronekit/dronekit-python #1009: Instantly Kill Throttle

rawillis98, 2020-03-12:
I need a way to instantly kill the throttle on the drone. I was going to use the RC Overrides but wondering if there's a better way?

Thanks in advance

peterbarker, 2020-03-13:
On Thu, 12 Mar 2020, Alex Willis wrote:

> I need a way to instantly kill the throttle on the drone. I was going to use the RC Overrides but wondering if there's a better way?

disarm the vehicle?  ArduPilot supports a "force" option to ensure that 
works.

You could also turn the "hardware safety switch" on via a mavlink command.

rawillis98, 2020-03-13:
Is there a dronekit function for this or do I need to generate the mavlink command myself?

I need to do it programatically in Python

peterbarker, 2020-03-14:
On Fri, 13 Mar 2020, Alex Willis wrote:

> Is there a dronekit function for this or do I need to generate the mavlink command myself?

I think you'll need to do it yourself.

kashifmaqsood, 2020-04-12:
> 
> 
> Is there a dronekit function for this or do I need to generate the mavlink command myself?
> 
> I need to do it programatically in Python

Flight mode "Brake" might help you there.
https://ardupilot.org/copter/docs/brake-mode.html#brake-mode