dronekit/dronekit-python #1016: Unique ID for AutoPilot and secure Mavlink Messages

shubham9436, 2020-04-04:
In the dronekit documentation, it is mentioned that
```It is *good practice* to assign a unique id for every system on the MAVLink network.

It is possible to configure the autopilot to only respond to guided-mode commands from a specified GCS ID.
```

Can someone tell me how to assign unique ID for a system and is there any provision to change that ID with authentication only?
I could see that there is an option for signing mavlink messages on https://mavlink.io/en/guide/message_signing.html but is there any implementation through dronekit?

mustafa-gokce, 2020-04-08:
I think it is related to "SYSID_THISMAV", "SYSID_MYGCS", "SYSID_ENFORCE".
Please refer to [https://ardupilot.org/copter/docs/parameters.html](https://ardupilot.org/copter/docs/parameters.html)
If you enforce the autopilot, from what I understand from docs, by setting "SYSID_ENFORCE", the vehicle only communicates to the control station which has an "SYSID_MYGCS" id. I couldn't find how to set ground control station id (dronekit connection id) while connecting to vehicle.

eminakgun, 2020-05-28:
It says assign different SYSID_THISMAV for autopilot you are trying to connect but dronekit assuming you are not receiving other system's messages from that port and you can not send messages to other systems like other UAVs around.
Here https://github.com/dronekit/dronekit-python/blob/master/dronekit/mavlink.py#L124, 
I guess source system must be different rather than 255, because I observed routing between ports TELEM1 and TELEM2 because autopilot has received a message with source system of 255 (actual GCS tho) from TELEM1. I gave the autopilot's SYSID_THISMAV to the source_system and 191 to source_component. But this only prevents MAV_CMD message routings I guess. Because they involve target_source and target_component fields.
I met another weird behavior of detecting wrong target system by dronekit, in the existince of routing,  so that my companion computer tried to pull parameters of another UAV. Because my telemetry radios are broadcasting.
So finally I forced dronekit to listen only the system which is given in parameter target_system in the constructor. And I dropped received packets from outer system.
Do you think I should create a PR for this? Idk if someone is still interested and looking at code reviews.