dronekit/dronekit-python #1017: execute more than one dronekit scipt at the same time

smiQi, 2020-04-07:
Hey guys,

is it possible to execute more than one dronekit script at the same time?

thanks and bye

mustafa-gokce, 2020-04-08:
Hi, @smiQi 
What do you want?
Do you want to control to the same vehicle using multiple scripts?
If so, yes.
You can connect to the vehicle using MAVProxy.
Then you can create multiple mavlink connection outputs from MAVProxy by assigning them individual ports.
Then you can connect these separate addresses from your individual scripts.
Also, you can do multiprocessing in a single script if you want to more things with your vehicle.
Please be clear.
Regards...

smiQi, 2020-04-09:
Hi @mustafa-gokce ,
thanks for your answer.

So i have i Raspberry Pi with the telemetry connect on my boat and want to host my own webserver on it with a lte stick.
So i want my own GUI build with a python flask server where i can start my scripts. I only need a few things like uploading the route to the ardupilot, start, stop and perhaps seeing all the parameters like gps coordinates and pitch,yaw...
I already have the server and can start the dronekit scripts one by one.
So my problem is, when i start the dronekit script for getting constantly the parameters i cant start another dronekit script like starting the boat.
Is this possible with mavproxy?
Sorry i'm new to this :)

Thanks already for your help!
Regards

mustafa-gokce, 2020-04-09:
@smiQi you can get multiple outputs from MAVProxy.
`mavproxy.py --master=/dev/serial0 --baudrate 921600 --aircraft MyCopter --out udp:127.0.0.1:14550 --out udp:127.0.0.1:14560`
You may need to configure the above MAVProxy command that suits with your serial connection.
Please look at the link:
https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html
Then you can connect from dronekit as  `connect("127.0.0.1:" + str(14550), wait_ready=True)` and also `connect("127.0.0.1:" + str(14560), wait_ready=True)`

smiQi, 2020-04-14:
@mustafa-gokce 
thanks, i will check this out :)

smiQi, 2020-04-16:
@mustafa-gokce 
thanks a lot!
It works perfectly!

closed 2020-04-16 by smiQi