dronekit/dronekit-python #1019: ERROR:dronekit.mavlink:Exception in MAVLink input loop

kevin-qing, 2020-04-11:
pi@raspberrypi:~/test $ python connect.py

WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1371, in listener
    self._heartbeat_error)
APIException: No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "connect.py", line 4, in <module>
    vehicle = connect('/dev/ttyUSB0', wait_ready=True, baud=921600)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 3166, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2275, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
----------------------------------------------------------------
code:
from dronekit import connect
 
# Connect to the Vehicle (in this case a UDP endpoint)
vehicle = connect('/dev/ttyUSB0', wait_ready=True, baud=921600)
 
# vehicle is an instance of the Vehicle class
print "Autopilot Firmware version: %s" % vehicle.version
print "Autopilot capabilities (supports ftp): %s" % vehicle.capabilities.ftp
print "Global Location: %s" % vehicle.location.global_frame
print "Global Location (relative altitude): %s" % vehicle.location.global_relative_frame
print "Local Location: %s" % vehicle.location.local_frame    #NED
print "Attitude: %s" % vehicle.attitude
print "Velocity: %s" % vehicle.velocity
print "GPS: %s" % vehicle.gps_0
print "Groundspeed: %s" % vehicle.groundspeed
print "Airspeed: %s" % vehicle.airspeed
print "Gimbal status: %s" % vehicle.gimbal
print "Battery: %s" % vehicle.battery
print "EKF OK?: %s" % vehicle.ekf_ok
print "Last Heartbeat: %s" % vehicle.last_heartbeat
print "Rangefinder: %s" % vehicle.rangefinder
print "Rangefinder distance: %s" % vehicle.rangefinder.distance
print "Rangefinder voltage: %s" % vehicle.rangefinder.voltage
print "Heading: %s" % vehicle.heading
print "Is Armable?: %s" % vehicle.is_armable
print "System status: %s" % vehicle.system_status.state
print "Mode: %s" % vehicle.mode.name    # settable
print "Armed: %s" % vehicle.armed    # settable

=======
raspi + px4

thank

mustafa-gokce, 2020-04-13:
Clearly you have a connection problem.
Check your wiring and configurations on the autopilot.
Also look at https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html
If you can not solve your problem, give some details next time.

eminakgun, 2020-06-06:
is your TELEM port connected to
> ttyUSB0
on raspi? or ttyAMA0?

chenge456, 2024-03-11:
I'm also having the same issue, have you solved it？Thank you!

harmor123, 2024-04-13:
I'm also having the same issue, have you solved it？Thank you!

harmor123, 2024-04-13:
[image]

harmor123, 2024-04-13:
python3:
```
from dronekit import connect

# Connect to the Vehicle (in this case a UDP endpoint)
vehicle = connect('/dev/ttyUSB0', wait_ready=True, baud=921600)

# vehicle is an instance of the Vehicle class
print("Autopilot Firmware version: %s" % vehicle.version)   #飞控版本
print("Autopilot capabilities (supports ftp): %s" % vehicle.capabilities.ftp) 
print("Global Location: %s" % vehicle.location.global_frame )    #全球定位信息（经纬度和高度）
print("Global Location (relative altitude): %s" % vehicle.location.global_relative_frame )  #相对坐标
print("Local Location: %s" % vehicle.location.local_frame)     #NED相对位置信息，东南西北
print("Attitude: %s" % vehicle.attitude)     #无人机朝向：roll,pitch,yaw单位弧度（-π~π）
print("Velocity: %s" % vehicle.velocity)     #三维速度（m/s）
print("GPS: %s" % vehicle.gps_0)             #GPS信息
print("Groundspeed: %s" % vehicle.groundspeed)          #地速（m/s）
print("Airspeed: %s" % vehicle.airspeed )               #空速（m/s）
print("Gimbal status: %s" % vehicle.gimbal )            #云台信息（单位度）
print("Battery: %s" % vehicle.battery      )            #电池信息
print("EKF OK?: %s" % vehicle.ekf_ok  )                 #扩展卡尔曼滤波器信息
print("Last Heartbeat: %s" % vehicle.last_heartbeat) 
print("Rangefinder: %s" % vehicle.rangefinder  )        #超声波或激光雷达模块状态
print("Rangefinder distance: %s" % vehicle.rangefinder.distance  )    #超声波或激光雷达模块的传感距离
print("Rangefinder voltage: %s" % vehicle.rangefinder.voltage)        #超声波或激光雷达模块电压
print("Heading: %s" % vehicle.heading )                               #无人机朝向（单位度）
print("Is Armable?: %s" % vehicle.is_armable      )                   #是否可以解锁
print("System status: %s" % vehicle.system_status.state)              #系统状态
print("Mode: %s" % vehicle.mode.name    )                             # settable飞行模式
print("Armed: %s" % vehicle.armed     )                               # settable解锁状态
```