dronekit/dronekit-python #1024: dronekit.APIException: mode (0, 4) not available on mavlink definition

bduva002, 2020-04-24:
Hello all, 
      I am running arduplane 4.1.X with dronekit 2.9.2 and pymavlink 2.4.4. When I run the SITL sim_vehicle.py and connect the vehicle_state.py example code every thing works as it should. 

Now when I run the vehicle_state.py on an RPI connected to a cube orange I get the following error message:

ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/dronekit/__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "/usr/local/lib/python3.5/dist-packages/dronekit/__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 4) not available on mavlink definition

From this error I see that from the real pixhawk m.custom_mode =0 and m.base_mode=4
Since the SITL is working I went into the dronekit __init__.py file and added some print statements to tell me what m.custom_mode and m.base_mode is when running with the SITL. 
When running dronekit connected to SITL:
m.custom_mode = 2
m.base_mode = 81

I am not sure why custom and base mode values are different when I run connected to a pixhawk cube orange.

I have tried building dronekit from source based on similar issues that are already on github but this did not make a difference. I know @peterbarker mentioned this might be a problem with multiple connections in this link https://github.com/dronekit/dronekit-python/issues/677 but I think this error is something different.

Any help would be greatly appreciated

bduva002, 2020-04-27:
Well I have an update, Thanks to the help of my good friend Tarik in this post:  
https://discuss.cubepilot.org/t/using-dronekit-with-cubepilot-orange-ads-b-carrier-board-exception-message-handler-error/3367
We are both using the new ADSB carrier boards with the pixhawk cube orange and ardupilot4.0.4. It turns out there must be cross talk because when we allow the ADSB receiver to be connected through serial5 to the pixhawk I get the following error message:
dronekit.APIException: mode (0, 4) not available on mavlink definition

If I disable SERIAL5_PROTOCAL=None the error goes away.

It sounds like these additional modes need to just be added to dronekit. Dose anyone out there have any ideas? 

Thanks

munsterlander, 2021-11-23:
Any progress on this?  I can't get anything going due to this error.  The Cube Orange can establish a connection, but I can't get passed is_armable as its always false and trying to set anything results in errors because the vehicle is never initialized.

```
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 0) not available on mavlink definition

Get all vehicle attribute values:
 Autopilot Firmware version: UnknownAutoPilotUnknownVehicleType18-4.1.0
   Major version number: 4
   Minor version number: 1
   Patch version number: 0
   Release type: rc
   Release version: 0
   Stable release?: True
 Autopilot capabilities
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 4) not available on mavlink definition   Supports MISSION_FLOAT message type: True

   Supports PARAM_FLOAT message type: True
   Supports MISSION_INT message type: True
   Supports COMMAND_INT message type: True
   Supports PARAM_UNION message type: False
   Supports ftp for file transfers: True
   Supports commanding attitude offboard: True
   Supports commanding position and velocity targets in local NED frame: True
   Supports set position + velocity targets in global scaled integers: True
   Supports terrain protocol / data handling: True
   Supports direct actuator control: False
   Supports the flight termination command: True
   Supports mission_float message type: True
   Supports onboard compass calibration: True
 Global Location: LocationGlobal:lat=0.0,lon=0.0,alt=None
 Global Location (relative altitude): LocationGlobalRelative:lat=0.0,lon=0.0,alt=-0.259
 Local Location: LocationLocal:north=None,east=None,down=None
 Attitude: Attitude:pitch=-0.01201585866510868,yaw=0.07953730970621109,roll=0.026009727269411087
 Velocity: [0.0, 0.0, 0.0]
 GPS: GPSInfo:fix=1,num_sat=0
 Gimbal status: Gimbal: pitch=None, roll=None, yaw=None
 Battery: Battery:voltage=14.991,current=0.0,level=99
 EKF OK?: False
 Last Heartbeat: 0.0
 Rangefinder: Rangefinder: distance=None, voltage=None
 Rangefinder distance: None
 Rangefinder voltage: None
 Heading: 4
 Is Armable?: False
 System status: STANDBY
 Groundspeed: 0.005084705539047718
 Airspeed: 0.0
 Mode: STABILIZE
 Armed: False
Basic pre-arm checks
 Waiting for vehicle to initialise...
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (33554831, 0) not available on mavlink definition
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
```

bduva002, 2021-11-24:
Are you using a cube orange?
Did you disable Serial5?
Have you gone through the calibrations such as accelerometer and compass cal

munsterlander, 2021-11-24:
Yes.  I set serial 5 protocol to -1. Reboot the board. Everything had been calibrated and tuned and flies fine. I have tried mav1 and mav2 as well. No change in the error. Running 4.1.1.

bduva002, 2021-11-25:
Lol before you throw it, I think the problem is the EKF is not happy. When you run this code are you outside and have a good gps fix. Also check mission planner. The hud will tell you if the EKF is not happy. 

Also have you tried commenting out the part that checks to see if the vehicle is armable?

When I started this post I could not get it to connect until I turned of the serial 5. The carrier does not seem to be the problem in your case.

munsterlander, 2021-11-25:
I should clarify, I can connect and I saw the EKF issue (battery as well). It's the constant heartbeat of the error above that ultimately is causing issues. I overrode the pre-arm checks to make sure I could move on and the script is fine.   I need to do some more tests and dump the logs here.  Hopefully I will have some time later.  Thank you for any pointers and help you have already provided. Your are like the only person responding so it's very much appreciated!

munsterlander, 2021-11-25:
Ok, I just ran it so you can see the script executes - regardless of the status - and the vehicle armed and did what I expected it to do.  What has been problematic, is sometimes you have to execute the script a few times as you get the timeout that is in the start of below which I attribute to the ADS-B issue that we can't seem to turn off.  In this current config for below, SERIAL5_PROTOCOL = None (-1), SERIAL2_PROTOCOL: Mavlink2, ADSB_ENABLE: Off:

```
dronekit.APIException: mode (33554837, 0) not available on mavlink definition
Traceback (most recent call last):
  File "TestTelemetry.py", line 18, in <module>
    vehicle.wait_ready('autopilot_version')
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 2375, in wait_ready
    timeout)
dronekit.TimeoutError: wait_ready experienced a timeout after 30 seconds.

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

C:\Users\munsterlander\PythonProjects\Drone>python TestTelemetry.py
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 0) not available on mavlink definition
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (33554837, 0) not available on mavlink definition
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 0) not available on mavlink definition
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (33554837, 0) not available on mavlink definition

Get all vehicle attribute values:
 Autopilot Firmware version: UnknownAutoPilotUnknownVehicleType0-4.1.1
   Major version number: 4
   Minor version number: 1
   Patch version number: 1
   Release type: rc
   Release version: 0
   Stable release?: True
 Autopilot capabilities
   Supports MISSION_FLOAT message type: True
   Supports PARAM_FLOAT message type: True
   Supports MISSION_INT message type: True
   Supports COMMAND_INT message type: True
   Supports PARAM_UNION message type: False
   Supports ftp for file transfers: True
   Supports commanding attitude offboard: True
   Supports commanding position and velocity targets in local NED frame: True
   Supports set position + velocity targets in global scaled integers: True
   Supports terrain protocol / data handling: True
   Supports direct actuator control: False
   Supports the flight termination command: True
   Supports mission_float message type: True
   Supports onboard compass calibration: True
 Global Location: LocationGlobal:lat=None,lon=None,alt=None
 Global Location (relative altitude): LocationGlobalRelative:lat=None,lon=None,alt=None
 Local Location: LocationLocal:north=None,east=None,down=None
 Attitude: Attitude:pitch=None,yaw=None,roll=None
[... 49 lines cut ...]
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 0) not available on mavlink definition
 Waiting for arming...
ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
[... cut ...]
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1531, in notify_message_listeners
    fn(self, name, msg)
  File "C:\Users\munsterlander\AppData\Local\Programs\Python\Python36-32\lib\site-packages\dronekit\__init__.py", line 1223, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
dronekit.APIException: mode (0, 0) not available on mavlink definition
Taking off!
Close vehicle
```

bduva002, 2021-11-25:
I have been using mavlink1 for my serial protocol.     You might check serial6 and turn it off too. The developers could have moved the ADSB port to another serial port. 

So if I am following you, your code will run with no error but when you run it again you get this error?

munsterlander, 2021-11-25:
I believe I have all the other ports set to off but I will check again.  

No on the second part. Basically, you have to run the script a few times before it fully runs. You get random timeouts. Like I can run it immediately after a good run and it will timeout.  It's weird.   As you can see, it does print out debug statements, it did arm the drone and did launch it,  but it's too unreliable for us to even demo this and because of that unpredictability, is just not safe to fly.   That's what I'm trying to solve by eliminating the error so we can hunt down other bugs or issues.

munsterlander, 2021-11-25:
So definitely not the right way of doing this, but this stopped all the errors, lol, and it stopped it from timing out as well.

Comment out lines 1250 and 1251 of https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L1250

This suppressed all the errors (big no no - I know) but our drone flew from the script and then proceeded to run itself into a wall due to me forgetting to turn on RTK and obstacle avoidance - oops.

bduva002, 2021-11-26:
Well there is more than one way to solve a problem. At least it's flying. Look out for those walls haha. You could add a check that looks at the rtk and obstacle avoidance Paramus in your dK script.

munsterlander, 2021-11-26:
So Alvin figured out part of the issue here as well: 
  https://discuss.cubepilot.org/t/connect-to-drone-via-herelink-and-drone-kit-python/7887/9?u=munsterlander

In short, it's time to move on from Dronekit.

bduva002, 2021-11-26:
Yeah, I wrote a whole data recorder with dK. It can be fical at times. I have started on the new lua script method. You might look into that. Lua scripts seem to work well so far.

munsterlander, 2021-11-26:
Definitely will look at the options out there.  I am using Python because our ground-based video system is running Python and so when it calculates lat/longs, it's easiest to just call another Python file and pass the values to the drone.  Either way, we will need to address the tech stack for the future, but I am going to do a Pymavlink test right now.  Thanks again for all your help and insight!  It was very much appreciated.

Captain299792458, 2022-07-02:
did anyone get a solve for this yet ??

angela804, 2022-08-09:
@munsterlander hello. Curious if you got past this.
I'm using HereLink controller, Cube Orange and connected my cube Orange to Raspberry Pi's Rx and Tx

pranav8112, 2024-11-01:
> @munsterlander hello. Curious if you got past this. I'm using HereLink controller, Cube Orange and connected my cube Orange to Raspberry Pi's Rx and Tx
hi there , did you solved that error ...? i am using pixhawk cube plus orange

munsterlander, 2024-11-02:
Sorry for missing the earlier posts. Yes, we did get passed it by commenting out those lines of code. Was it ideal, no, but it did work.  We moved into a different platform, so I don't have much more than what was originally posted.

angela804, 2024-11-05:
> > @munsterlander hello. Curious if you got past this. I'm using HereLink controller, Cube Orange and connected my cube Orange to Raspberry Pi's Rx and Tx
> > hi there , did you solved that error ...? i am using pixhawk cube plus orange

don't know what's up with that warning. It's weird but disregarding the warning, I was still able to connect to the pixhawk and control the robot using the connected raspberry pi.
you can find the source code for that and comment it so that it won't always appear.
Moving to a different platform that what I previously used, that did not appear

hamishwillee, 2025-01-19:
This will map to https://mavlink.io/en/messages/common.html#SET_MODE or possibly https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_MODE "under the hood". The base mode is I believe largely ignored and the custom mode is flight stack specific.
 
You can look at the source code in Pymavlink mavutil to determine the mode mappings for both ArduPilot and PX4. For recent PX4 you could dynamically determine the mapping using the standard modes API (But not through dronekit, since it is too out of date).

ryochiji, 2026-04-03:
In case any other stubborn old souls trying to keep their DroneKit-based code alive stumbles on this, this issue is likely a duplicate of issue #1187 which was fixed in #1188.