dronekit/dronekit-python #1025: how to instantly observe throttle% parameter?

Guohao91, 2020-05-11:
[image]
 hi guys! 
A little question, can dronekit do the observation of "Thr...%" as the Mavproxy does?
I just need the thr% para now.
Thanks in advance!!!!

bduva002, 2020-05-20:
So throttle percentage is in the VFR_HUD Mavlink message.
https://mavlink.io/en/messages/common.html#VFR_HUD
Dronekit dose contain the VFR_HUD message here:
https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L1094
But they did not write out all the attributes of the VFR_HUD message ie Throttle %
I recommend take a look at the create attribute example in the dronekit example folder. 
https://github.com/dronekit/dronekit-python/tree/master/examples

Guohao91, 2020-05-20:
@bduva002 I did go through all the examples, i even tried the para. "vehicle.parameters." and "vehicle.throttle()", nothing worked yet to me. 
Anyways thanks for your advice.

bduva002, 2020-05-20:
It should be vehicle.throttle not vehicle.throttle()

Guohao91, 2020-05-20:
@bduva002 doesn't work neither.

Guohao91, 2020-05-26:
@bduva002 hi, bru. I've tried the "VFD_HUD" message content through the example frame of "create_attribute.py".
Like in (my_vehicle.py):

from dronekit import Vehicle

class VFRHUD(object):

    def __init__(self, airspeed=0, groundspeed=0, heading=0, throttle=0):
        """
        RawIMU object constructor.
        """
        self.airspeed = airspeed
        self.groundspeed = groundspeed
        self.heading = heading
        self.throttle = throttle  
        
    def __str__(self):
        """
        String representation used to print the RawIMU object. 
        """
        return "VFR_HUD: airspeed={},groundspeed={},heading={},throttle={}".format(self.airspeed, self.groundspeed, self.heading,self.throttle)

class MyVehicle(Vehicle):
    def __init__(self, *args):
        super(MyVehicle, self).__init__(*args)

        # Create an Vehicle.raw_imu object with initial values set to None.
        self._vfr_hud = VFRHUD()

        # Create a message listener using the decorator.   
        @self.on_message('VFR_HUD')
        def listener(self, name, message):
            """
            The listener is called for messages that contain the string specified in the decorator,
            passing the vehicle, message name, and the message.
            
            The listener writes the message to the (newly attached) ``vehicle.raw_imu`` object 
            and notifies observers.
            """
            self._vfr_hud.airspeed=message.airspeed
            self._vfr_hud.groundspeed=message.groundspeed
            self._vfr_hud.heading=message.heading
            self._vfr_hud.throttle=message.throttle
            
            # Notify all observers of new message (with new value)
            #   Note that argument `cache=False` by default so listeners
            #   are updated with every new message
            self.notify_attribute_listeners('vfr_hud', self._vfr_hud) 

    @property
    def vfr_hud(self):
        return self._vfr_hud

And in (creat_attribute.py):

if not connection_string:
    connection_string = "/dev/ttyACM0"

print('Connecting to vehicle on: %s' % connection_string)
vehicle = connect(connection_string, wait_ready=True, vehicle_class=MyVehicle,baud=921600)

def vfd_hud_callback(self, attr_name, value):
    # attr_name == 'raw_imu'
    # value == vehicle.raw_imu
    print(value)

vehicle.add_attribute_listener('vfd_hud', vfd_hud_callback)

print('Display RAW_IMU messages for 5 seconds and then exit.')
time.sleep(5)

#The message listener can be unset using ``vehicle.remove_message_listener``

#Close vehicle object before exiting script
print("Close vehicle object")
vehicle.close()

I ran them, there is no problem came out, however no expected information displayed either.
[image]
Can you find out why? Huge appreciate it!

bduva002, 2020-05-26:
try removing these lines

def vfd_hud_callback(self, attr_name, value):
# attr_name == 'raw_imu'
# value == vehicle.raw_imu
print(value)

vehicle.add_attribute_listener('vfd_hud', vfd_hud_callback)

print('Display RAW_IMU messages for 5 seconds and then exit.')
time.sleep(5)

#The message listener can be unset using vehicle.remove_message_listener

Replace with

while True:
    print(vehicle.vfr_hud.throttle)
    time.sleep(.1)

Guohao91, 2020-05-27:
@bduva002  Woww, your answer makes it!
You are a real master!!

closed 2020-05-27 by Guohao91

bduva002, 2020-05-27:
Glad it worked!

also I think your other way may have worked too but i think you have a typo
here

vehicle.add_attribute_listener('vfd_hud', vfd_hud_callback)

I think vfd_hud should be vfr_hud
You could try it

also if you just say
print(vehicle.vfr_hud)

this should print out all attributes related to vfr_hud as you typed them
here

"VFR_HUD: airspeed={},groundspeed={},heading={},throttle={}".format(self.airspeed,
self.groundspeed, self.heading,self.throttle)

I normally will spell out only the one I want for example

throttle = vehicle.vfr_hud.throttle

this way I don't have to keep writing out vehicle.vfr_hud.throttle

Guohao91, 2020-05-27:
@bduva002  
oh dear god!  That is definitely my typo! Thanks bru!
And you are right, vehicle.vfr_hud can print out all the attributes, I just need the throttle para. in my app.
Jesus, i have a typo in there, I can believe it!

bduva002, 2020-05-27:
yea typo are easy to do 

also fyi you can just keep adding to you my_vehicle.py in the same format for any other Mavlink message donekit dose not already have.

Guohao91, 2020-05-27:
Sure it can do. thanks again~!

Guohao91, 2020-05-27:
@bduva002  HI, bru I also have a small question regarding the example (gcs.py)
[https://github.com/dronekit/dronekit-python/blob/master/examples/gcs/microgcs.py](url)
[image]
so after the posted attitude degrees, how to add up degree symbol in its sentence (line 63)??
(I know it is actually radians value).

bduva002, 2020-05-27:
you mean how to convert attitude to degrees?

import math

then add the conversion ie
pitch = vehicle.attitude.(attribute for pitch gose here)*180/3.14

Guohao91, 2020-05-27:
@bduva002 hahaha no, I mean just how to add up ° character.
Plz forget about it. I figured that out. Thanks!

bduva002, 2020-05-27:
haha okay no worries

Guohao91, 2020-06-01:
@bduva002  yo bruuuu!
here is another question~~  did you ever see there is a function "condition_yaw(heading, relative=False):" in the example
 [https://github.com/dronekit/dronekit-python/blob/master/examples/guided_set_speed_yaw/guided_set_speed_yaw.py](url)
My question is if the raw command "mavutil.mavlink.MAV_CMD_CONDITION_YAW" can be used under NoGPS mode? 
looking forward to your answer~~

bduva002, 2020-06-02:
I have not tried that but assume yes as it probably uses the compass for
the heading. I would recommend trying it in the SITL

Guohao91, 2020-06-03:
@bduva002  thanks bru. But no, unfortunately the command codition-yaw can't work under Guided_NoGPS, I've checked them out. I guess I must use set_attitude method to do rotation as good as possible.