dronekit/dronekit-python #1032: Dronekit timeout, no heartbeat in last 5 seconds

ktsintotas, 2020-06-08:
_While using the command:_

**edison@jubilinux:~$ mavproxy.py --master=/dev/ttyMFD1 --baudrate=921600**

_I get the following response from the system meaning that my companion-computer communicates correctly with the flight controller: -->_ 

**Connect /dev/ttyMFD1 source_system=255
Failed to load module: No module named terrain. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyMFD1
 MAV> online system 1
RTL> Mode RTL
MAV> Received 1039 parameters
Saved 1039 parameters to mav.parm**

_However, when I try to connect through dronekit using commands:_

**edison@jubilinux:~$ python
from dronekit import connect
vehicle = connect('/dev/ttyMFD1', wait_ready=True, baud=921600)**

_this error occurs: -->_

**WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1371, in listener
    self._heartbeat_error)
APIException: No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 3172, in connect
    timeout=timeout)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2375, in wait_ready
    timeout)
dronekit.TimeoutError: wait_ready experienced a timeout after 30 seconds.**

===========================================

_My system consists of the bellow materials:_

Flight controller: Pixhawk 2.0 (Cube black)
Flight controller firmware: ArduPlane 4.0.5

Companion computer: Intel Edison
Companion computer firmware: Jubilinux

DroneKit version: 2.9.2
Pymavlink version: 2.4.8
MAVproxy version: 1.8.19
 
Any opinion?

ktsintotas, 2020-06-29:
The reason behind this issue is the ardupilot firmware "arduplane". Since I tried the same process using Rasberry pi, the companion computer presented the same communication error. Even if older firmware and beta versions were used, the problem remains, while utilizing the copter firmware the communication between companion computer (Edison, Rasberry pi 3 B) works normally. So an open issue remains for "arduplane".

ktsintotas, 2020-08-24:
Thanks for the info @eminakgun! Another question: which version of python you used?

I ask this because I observed than pymavlink works more robust in python 3.

eminakgun, 2020-08-24:
I work with python 2.7. I plan to move on to python 3

cherrywalter, 2021-03-02:
I have the exact same issue. I can run _**mavproxy.py --master=/dev/ttyAMA0**_ and can even arm the drone with _arm throttle_ .
But when i try to run a python script, it gives me the same error:
**WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds**

My setup:
Pixhawk Cube Black with Arducopter v4.07
Pixhawk with Pi OS Lite

Python Version: 2.7.16
PyMavlink: 2.4.14
MavProxy: 1.8.34

Has anyone had the same issue and figured this out?

ktsintotas, 2021-03-03:
> I have the exact same issue. I can run _**mavproxy.py --master=/dev/ttyAMA0**_ and can even arm the drone with _arm throttle_ .
> But when i try to run a python script, it gives me the same error: [...]

Try to use Python 3.

EngDavidVella, 2022-03-25:
Has any one ever sorted this out?

I have the same issue with python 3.9.
Everything else works fine.