dronekit/dronekit-python #1037: mission planner mavproxy telemetry radio unable to connect dronekit

DoneMan, 2020-06-21:
hi all

i’m trying to connect mavproxy, mission planner and dronekit to my hexa.
i am not using a companion computer my setup is as follows:
i have a pc running windows for mission planner and on that same pc im running a virtual machine with ubuntu 16.04 running dronekit and mavproxy.

first i’m running mavproxy and forking out 2 connections, 1 for mission planner and 1 for dronekit with the following command: **mavproxy.py --master=/dev/ttyUSB0(im telling mavproxy to use the serial which the telemetry dongle is connected to) --baudrate=57600 --out=127.0.0.1:14551(ip adress on my virtual ubuntu) --out=10.0.0.7:14550 (ip for my windows machine)**. i am able to connect to the drone in mission planner via the udp at port 14550. and i am able to send commands via mavproxy console such as change modes/arm etc.

my problem is when i execute my connect script from dronekit to connect to the drone im getting a timeout error.
**the code:

from dronekit import connect, VehicleMode, LocationGlobalRelative
import time
import argparse

parser = argparse.ArgumentParser(description = ‘commands’)
parser.add_argument(’–connect’)
args =parser.parse_args()

connection_string = args.connect
vehicle = connect(connection_string ,wait_ready=True)

when i open a new terminal i run the command to launch and connect drone kit : python connect.py --connect 127.0.0.1:14551**
once i run this it gets stuck and after 30 seconds i get the following output:

dronekit.TimeoutError: wait_ready experienced a timeout after 30 seconds.

sometimes i get the following output:
gcs failsafe cleared

and then it gets stuck and after 30 seconds:

dronekit.APIException: Timeout in initializing connection.

i searched for about 3 days for solutions and tried everything that is out there pretty much, such as telling dronekit to wait longer than 30 seconds, but still i get the same error.
i updated dronekit to latest version updated pymavlink and pyserial. also downgraded them to check if it will solve this but nothing works.

**my main question is: is this error happening because dronekit can only run on a companion computer? and is not built for connecting threw telemetry radio?
if not, what am i doing wrong?**

i sincerely am lost and praying that someone can help me.
thanks in advanced. i would literally pay someone for a solution.

and again, sorry for the double post.

RomulusSam, 2020-08-07:
I feel your pain, I have the same problem and can't find the solution online. Have you figured this out, or do you have any new insights?

DoneMan, 2020-08-07:
hi, I actually solved it by changing my hardware. I changed my radio telemetry to something with a larger bandwidth. The fact is that the simple 3dr style radio telemetry are, pardon my French, shit. 
Change your telemtry radio to sth with a larger bandwidth  may i suggest xbee radio pro. 
And change this line:
vehicle = connect(connection_string ,wait_ready=True)
To:
vehicle = connect(connection_string ,wait_ready=False)

The wait_ready just means it will download the parameters from the vehicle. In case you absolutely need them, leave it on true and get a better radio system.

Hope this helps