dronekit/dronekit-python #1041: SITL test is failing: copter mode stuck on STABILIZE

pietrodn, 2020-07-04:
The SITL tests are currently failing on the master branch, both for [Python 2](https://app.circleci.com/pipelines/github/dronekit/dronekit-python/1/workflows/05098a02-aef1-4dc0-94ae-c93331c254a4/jobs/1418/steps) and [Python 3](https://app.circleci.com/pipelines/github/dronekit/dronekit-python/1/workflows/05098a02-aef1-4dc0-94ae-c93331c254a4/jobs/1419/steps).
The cause is a SITL test that's not passing: [test_goto.py](https://github.com/dronekit/dronekit-python/blob/master/dronekit/test/sitl/test_goto.py).

The apparent reason of the error is that the test times out when setting the mode to `GUIDED`.
I managed to run the unit test locally, and I can't get the SITL to switch to `GUIDED`: the vehicle remains stuck on `STABILIZE`. The GPS and the global location seem to be ok. The Vehicle is periodically receiving heartbeat updates containing the mode flag, so it's not a reception problem.

My hypothesis is that some update on the ArduPilot side, or in some dependencies, made the tests fail. From the [commit history] is easy to see that the previous commits (that worked) are identical to the latest ones, which were simple documentation changes.

Any hints?

pietrodn, 2020-07-04:
The problem is in `pymavlink`. The latest passing tests use pymavlink 2.4.0, and with that version the test passes.
I ran a bisect and pymavlink is OK until 2.4.8: the 2.4.9 release broke something.

pietrodn, 2020-07-04:
Found the problematic commit: https://github.com/ArduPilot/pymavlink/commit/1fa59ff0bbc0999353b30feac3afe5cfd67c08a9

peterbarker, 2020-07-04:
On Sat, 4 Jul 2020, Pietro De Nicolao wrote:

> Found the problematic commit: ArduPilot/pymavlink@1fa59ff

Looks like my bad.  What version of ardupilot's being run in SITL by 
default?!

I tend to run master ArduPilot when I'm using dronekit-python.

Fix here is probably to have dronekit-sitl pull a binary from ArduPilot's 
firmware.ardupilot.org; I have a branch here which does that.

Peter

pietrodn, 2020-07-05:
By default dronekit-sitl runs ArduPilot Copter v3.3. Probably it’s better to test against an updated firmware version as you said 👍

peterbarker, 2020-07-06:
On Sun, 5 Jul 2020, Pietro De Nicolao wrote:

> By default dronekit-sitl runs ArduPilot Copter v3.3. Probably it’s better to test against an updated firmware version as you said 👍

https://github.com/dronekit/dronekit-sitl/pull/145

So that kinda-sorta works.  At least far enough for the simple_goto.py 
dronekit-python example to work with no extra arguments.

Really not happy with the parameters stuff, obviously.

sanky1234, 2020-07-18:
Hi is there fix to this? I have been trying all sorts of things but havent been able to run the simple_goto.py.

rmccarth, 2020-07-20:
Hi, in the meantime should we revert to using 2.9.1? (I noticed there's a 2 year gap on pypi)

sanky1234, 2020-07-20:
> Hi, in the meantime should we revert to using 2.9.1? (I noticed there's a 2 year gap on pypi)
> 
> Nevermind. I was able to uninstall pymavlink with
> 
> ```
> pip3 uninstall pymavlink
> ```
> 
> and then reinstall the older version with
> 
> ```
> pip3 install pymavlink==2.4.8
> ```

Did this work for you by just reinstalling an old version of pymavlink? It wasn’t working for me.

rmccarth, 2020-07-20:
> Did this work for you by just reinstalling an old version of pymavlink? It wasn’t working for me.

Yeah reverting pymavlink to 2.4.8 did the trick for me. If you are having trouble in general I would suggest running the dronekit-python/examples/vehicle_state/vehicle_state.py to check if there is a bug in your code.

punkypankaj, 2020-11-08:
> Yeah reverting pymavlink to 2.4.8 did the trick for me. If you are having trouble in general I would suggest running the dronekit-python/examples/vehicle_state/vehicle_state.py to check if there is [...]

yes this reverting back to older version of pymavlink worked for me, i was having trouble in DO_SET_MODE change while executing the python script as it was stuck at stabalize, even now i can't change the mode in mavproxy but it works when i run the script

i was having pymavlink 2.4.12 and 
reverted back to pymavkink 2.4.8
for now at least im able to run the script and need to find the issue with mavproxy

walchko, 2021-01-09:
pymavlink 2.4.14 is junk also ... reverting to 2.4.8 fixed it

danammeansbear, 2021-01-14:
the fix doesnt work at this point, just ends up spitting a bunch of horse crap on my screen.

danammeansbear, 2021-01-14:
'''
AttributeError: 'MAVLink_vibration_message' object has no attribute '_instance_field'
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\dronekit\mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "C:\ProgramData\Anaconda3\lib\site-packages\dronekit\__init__.py", line 1370, in listener
    raise APIException('No heartbeat in %s seconds, aborting.' %
dronekit.APIException: No heartbeat in 30 seconds, aborting.
ERROR:dronekit.mavlink:Exception while receiving message: 
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\dronekit\mavlink.py", line 218, in mavlink_thread_in
    msg = self.master.recv_msg()
  File "C:\ProgramData\Anaconda3\lib\site-packages\pymavlink\mavutil.py", line 456, in recv_msg
    return None
  File "C:\ProgramData\Anaconda3\lib\site-packages\pymavlink\mavutil.py", line 372, in post_message
  File "C:\ProgramData\Anaconda3\lib\site-packages\pymavlink\mavutil.py", line 83, in add_message
    '''
AttributeError: 'MAVLink_heartbeat_message' object has no attribute '_instance_field'
ERROR:dronekit.mavlink:Exception while receiving message: 
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\dronekit\mavlink.py", line 218, in mavlink_thread_in
    msg = self.master.recv_msg()
  File "C:\ProgramData\Anaconda3\lib\site-packages\pymavlink\mavutil.py", line 456, in recv_msg
    return None
  File "C:\ProgramData\Anaconda3\lib\site-packages\pymavlink\mavutil.py", line 372, in post_message
  File "C:\ProgramData\Anaconda3\lib\site-packages\pymavlink\mavutil.py", line 83, in add_message
    '''
AttributeError: 'MAVLink_heartbeat_message' object has no attribute '_instance_field'
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\dronekit\mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "C:\ProgramData\Anaconda3\lib\site-packages\dronekit\__init__.py", line 1370, in listener
    raise APIException('No heartbeat in %s seconds, aborting.' %
dronekit.APIException: No heartbeat in 30 seconds, aborting.

Mettikodeva-Nara, 2021-11-03:
> By default dronekit-sitl runs ArduPilot Copter v3.3. Probably it’s better to test against an updated firmware version as you said 👍

how to run dronekit-sitl with updated firmware version?
thanks

Xarbenence, 2023-01-21:
I've been experiencing the same issue of telling my SITL to enter GUIDED mode, but it gets stuck in the loop waiting for the mode to change. I tried to downgrade to pymavlink==2.4.8, but I'm getting a legacy-install-failure. Has anyone experienced this?
I'm on a Mac with an M1 chip, if that helps. I've also noticed that downgrading about anything via pip usually ends in an error, but not necessarily the same error. 

alexsharper@dhcp-10-105-11-95 ~ % sudo pip3 install pymavlink==2.4.8
WARNING: The directory '/Users/alexsharper/Library/Caches/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Collecting pymavlink==2.4.8
  Downloading pymavlink-2.4.8.tar.gz (3.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.6/3.6 MB 28.4 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: future in /opt/homebrew/lib/python3.9/site-packages (from pymavlink==2.4.8) (0.18.3)
Requirement already satisfied: lxml in /opt/homebrew/lib/python3.9/site-packages (from pymavlink==2.4.8) (4.9.2)
Building wheels for collected packages: pymavlink
  Building wheel for pymavlink (setup.py) ... error
  error: subprocess-exited-with-error
  
  **× python setup.py bdist_wheel did not run successfully.
  │ exit code: 1**
  ╰─> [54 lines of output]
      running bdist_wheel
      running build
      running build_py
      Using message definitions from /private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/message_definitions
      Building /private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/message_definitions/v1.0/python_array_test.xml for protocol 1.0
      WARNING: Unable to load XML validator libraries. XML validation will not be performed
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/setup.py", line 109, in <module>
          setup (name = 'pymavlink',
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/__init__.py", line 87, in setup
          return distutils.core.setup(**attrs)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 148, in setup
          return run_commands(dist)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
          dist.run_commands()
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
          self.run_command(cmd)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/dist.py", line 1229, in run_command
          super().run_command(command)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/opt/homebrew/lib/python3.9/site-packages/wheel/bdist_wheel.py", line 299, in run
          self.run_command('build')
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/dist.py", line 1229, in run_command
          super().run_command(command)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/command/build.py", line 136, in run
          self.run_command(cmd_name)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/dist.py", line 1229, in run_command
          super().run_command(command)
        File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/setup.py", line 103, in run
          generate_content()
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/setup.py", line 64, in generate_content
          if not mavgen.mavgen_python_dialect(dialect, mavparse.PROTOCOL_1_0):
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/generator/mavgen.py", line 247, in mavgen_python_dialect
          if not mavgen(opts, [xml]):
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/generator/mavgen.py", line 142, in mavgen
          xml.append(mavparse.MAVXML(fname, opts.wire_protocol))
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/generator/mavparse.py", line 420, in __init__
          m.crc_extra = message_checksum(m)
[... cut ...]
        File "/private/tmp/pip-install-qkyxr4il/pymavlink_7b139926334f4849b671942012c96389/generator/mavcrc.py", line 34, in accumulate_str
          bytes.fromstring(buf)
      AttributeError: 'array.array' object has no attribute 'fromstring'
      [end of output]
  
  **note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure**

× Encountered error while trying to install package.
╰─> pymavlink

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

Miles5177, 2023-03-03:
Have you been able to find a solution to this? thank you!

mujtabafromthedarkside, 2023-04-04:
I am also getting the same as above legacy-install-failure as above with Python3. I am on Windows 10. The installation works fine for python2.

danammeansbear, 2023-04-05:
i am having trouble on multiple systems. cant seem to figure out how to get
it to work on ubuntu or my local windows machine which had it working a
year ago but now its caput and im tooo scared to test it on my drone
because i need to demo it for my senior project.
*Adam Dabdoub*
Software Developer
c: 513.886.0301

SarinleFreeman, 2023-04-05:
> i am having trouble on multiple systems. cant seem to figure out how to get it to work on ubuntu or my local windows machine which had it working a year ago but now its caput and im tooo scared to [...]

It works on Ubuntu for me, perhaps set up an associated Docker Image which uses Ubuntu and run your code on there?

Otherwise I can set up one and I'll link it in this chat.

danammeansbear, 2023-04-05:
I think it would be nice to have an official docker image that we can use
as a base so there less hiccups in getting into development because as it
stands, the tutorial is out of date if legacy users are having issues just
getting back into the code base.
*Adam Dabdoub*
Software Developer
c: 513.886.0301

Atigit15, 2023-04-06:
>The installation works fine for python2

Can you tell me how to install the desired version of pymavlink for python2, since pip is no longer supported for python2. Thank you !

wangcongai, 2024-05-15:
After pymavlink version==2.4.8, the set_mode_apm function in pymavlik.mavutil.py changed, which cause failure on ardupilot MODE CHANGE.
In order to use new version like pymavlink version==2.4.41, you just need to comment the code on "mav.command_long_send" and change it back to old "mav.set_mode_send". Reboot your IDE, Then the problem solved!
[image]

mujtabafromthedarkside, 2024-08-11:
> After pymavlink version==2.4.8, the set_mode_apm function in pymavlik.mavutil.py changed, which cause failure on ardupilot MODE CHANGE. In order to use new version like pymavlink version==2.4.41, you just need to comment the code on "mav.command_long_send" and change it back to old "mav.set_mode_send". Reboot your IDE, Then the problem solved! <img alt="image" width="1308" src="https://private-user-images.githubusercontent.com/143485711/330699021-8543640a-147a-4868-adda-a96b55fb6803.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMzOTI0MjAsIm5iZiI6MTcyMzM5MjEyMCwicGF0aCI6Ii8xNDM0ODU3MTEvMzMwNjk5MDIxLTg1NDM2NDBhLTE0N2EtNDg2OC1hZGRhLWE5NmI1NWZiNjgwMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxMVQxNjAyMDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNmM0MTMyN2IyNjVhODc5MWFmMDExMWUyYmQxOGUzOGM1ODVkMjg4N2ZhMTdhYmEzOTY2MTU4Y2RjNzBkYzc4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HgXv2fy0QX7xz48xZPLWuuyeIl6N4BmysDA1QE0jXdM">

Yeah, that works. I guess, this is a good way to make it run on latest python versions that do not support pymavlink version 2.4.8

Previously, I used to use either python 3.6 or python 2.7 which were compatible with pymavlink==2.4.8