dronekit/dronekit-python #1044: Guided Mode Movement and Commands for FIXED-WINGS

Ayvazof, 2020-07-24:
The issue is that, that I need to control plane from script depending of external sensor only way I found is use RC OVERRIDE but it is not a good way how I understand. Is it a way do it in right way?
Thanks in advance

myusf01, 2020-10-22:
I've been looking for the same but I couldn't found any related documentation about to set yaw, pitch, roll like commands like in [``send_ned_velocity``](https://github.com/dronekit/dronekit-python/blob/master/docs/guide/copter/guided_mode.rst#velocity-control) function or [``CMD_CONDITION_YAW``](https://github.com/dronekit/dronekit-python/blob/master/docs/guide/copter/guided_mode.rst#velocity-control). RC_OVERRIDE looks complicated and dangerous as far as I've seen in issues.

HefnySco, 2020-10-22:
You can use cruise, fbwa or fbwb with channel override. These modes will
limit the max pitch and roll while gives you good navigation control.

domnantas, 2021-03-13:
This was a wild ride, but I managed to figure it out.
First off, [Ardupilot documentation](https://ardupilot.org/dev/docs/plane-commands-in-guided-mode.html) lists `MAV_CMD_NAV_WAYPOINT` as the _only_ message available for planes. Apparently that's not true, you can also use `SET_ATTITUDE_TARGET`! Dronekit does not mention that anywhere either. In fact, in [set_attitude_target example](https://github.com/dronekit/dronekit-python/blob/c7143e0ba94d414416fd36679fd3306ce31a7e69/examples/set_attitude_target/set_attitude_target.py) it says *(Copter Only)*. This was really confusing.

After browsing through some similar issues in this repo, I decided to try it out and it worked! Here's my implementation:
```py
def send_set_attitude_target(roll=0, pitch=0, yaw=0, thrust=0.5):
    attitude = [np.radians(roll), np.radians(pitch), np.radians(yaw)]
    attitude_quaternion = quaternion.QuaternionBase(attitude)

    msg = vehicle.message_factory.set_attitude_target_encode(
        0, 0, 0,  # time_boot_ms, target_system, target_component
        0b10111000,  # type_mask https://mavlink.io/en/messages/common.html#ATTITUDE_TARGET_TYPEMASK
        attitude_quaternion,  # Attitude quaternion
        0, 0, 0,  # Rotation rates (ignored)
        thrust  # Between 0.0 and 1.0
    )

    vehicle.send_mavlink(msg)
```