dronekit/dronekit-python #1045: RTK inject possible?

FedePeralta, 2020-07-28:
Is there a way to inject RTK into the vehicle through dronekit?
I've found that the current version locks only with one of 4 fixes (fix_type (Int) – 0-1: no fix, 2: 2D fix, 3: 3D fix) https://dronekit-python.readthedocs.io/en/latest/automodule.html#dronekit.GPSInfo but some GPS like Here+ can inject info from external modules (https://ardupilot.org/copter/docs/common-here-plus-gps.html)
If not, what's the general road map in order to include RTK Compatibility?

FedePeralta, 2020-09-18:
If someone finds it useful, this code is the adaptation from mission planners gps_inject function, and it works
```
def rtk_comm(self):
        sleep(self.SLEEP_CONNECT)
        with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
            s.connect((self.rtk_base_port.split(':')[0], int(self.rtk_base_port.split(':')[1])))
            while True:
                rec_data = s.recv(1024)
                self.inject_seq_no += 1
                l_rec_data = len(rec_data)
                data = [0b0 for i in range(180)]
                gps = dict()  # mavlink_gps_rtcm_data_t gps = new mavlink_gps_rtcm_data_t();
                msglen = 180
                if l_rec_data > 720:
                    continue
                nopackets = l_rec_data / msglen + 1 if ((l_rec_data % msglen) == 0) else (l_rec_data / msglen) + 1
                nopackets = np.floor(nopackets).astype(np.int64)
                if nopackets >= 4:
                    nopackets = 4
                for a in range(nopackets):
                    if nopackets > 1:
                        gps["flags"] = 1
                    else:
                        gps["flags"] = 0
                    gps["flags"] += ((a & 0x3) << 1)
                    gps["flags"] += ((self.inject_seq_no & 0x1f) << 3)
                    _copy = np.minimum(l_rec_data - a * msglen, msglen)
                    gps["data"] = deepcopy(data)
                    for i in range(_copy):
                        gps["data"][i] = rec_data[a * msglen + i]
                    gps["len"] = _copy
                    aux_gps = self.vehicle.message_factory.gps_rtcm_data_encode(
                        gps["flags"],
                        gps["len"],
                        gps["data"]
                    )
                    self.vehicle.send_mavlink(aux_gps)
                self.vehicle.flush()
```

closed 2020-09-18 by FedePeralta

WCKBEN, 2023-08-04:
> If someone finds it useful, this code is the adaptation from mission planners gps_inject function, and it works
>
> ```
> def rtk_comm(self):
>         sleep(self.SLEEP_CONNECT)
>         with [...]