dronekit/dronekit-python #1047: wait_ready experienced an error after 30 seconds

RomulusSam, 2020-08-07:
Hi dronekit!

I'm trying (and failing) to connect to my drone through ttyTHS1, and I'm pretty sure that's the correct port since I tried ttyTHS2 and instantly wasn't able to connect. I'm baffled by this error since last week I was able to connect just fine using the same commands I am using now: vehicle = connect('/dev/ttyTHS1', wait_ready=True, baud=57600). Whenever I try to connect, after a short period of nothing, I am given the error "dronekit.TimeoutError wait_ready experienced an error after 30 seconds".

Is there anything else that I didn't check that I don't know about? I'm lost at sea here and could use any help anyone has to offer. Let me know if you need any other info too and I can supply it.

Much appreciated,
Sam

amgaber95, 2020-08-30:
I have the same issue, but couldn't solve the problem

ahmetozkaraca, 2020-10-23:
ı have the same problem, Can anyone help?