dronekit/dronekit-python #1048: connect() hangs with no info why on Cube Orange

siadajpan, 2020-08-17:
I'm running dronekit on NVIDIA Jetson Nano connecting through UART with Pixhawk Cube Orange.
dronekit version = 2.9.2
pymavlink version = 2.4.9
python version = 3.6.9

After running ```mavproxy.py --master=/dev/ttyTHS1 --baudrate=115200```
I'm getting:
```
Connect /dev/ttyTHS1 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyTHS1
MAV> online system 1
STABILIZE> Mode STABILIZE
```
but after running 
```
>>> import dronekit
>>> v = dronekit.connect('/dev/ttyTHS1', baud=115200)
```
It hangs without showing anything more (not even heartbeat timeout), so it seems like heartbeat is present.
I can see the some incoming messages every second after running ```screen /dev/ttyTHS1```

lachlanbrewster, 2020-08-18:
Try a different baud rate? Check what the Pixhawk is expecting in Mission planner and use that rate, but default is 57600 I think.

siadajpan, 2020-08-18:
I've changed baudrate in mission planner to 115200 and protocol to Mavlink2. I'm also using the same baudrate in mavproxy.py run

athoca, 2021-04-29:
Hi @siadajpan , do you find solution? I have the same issue but on Jetson Xavier. When connecting using mavproxy.py on Xavier, it shows only:
mavproxy.py --master=/dev/ttyTHS0 --baudrate 921600
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect /dev/ttyTHS0 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyTHS0
MAV> online system 1
ALT_HOLD> Mode ALT_HOLD

While connecting using mavproxy.py on Ground Computer, it shows more information:
mavproxy.py --master=/dev/cu.usbserial-AD0K5Q02
Connect /dev/cu.usbserial-AD0K5Q02 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/cu.usbserial-AD0K5Q02
MAV> GPS lock at 0 meters
AP: GCS Failsafe Cleared
online system 1
ALT_HOLD> Mode ALT_HOLD
AP: PreArm: Radio failsafe on
AP: ArduCopter V4.0.4 (40502bd9)
AP: ChibiOS: d4fce84e
AP: CubeOrange 0026003A 34385111 31373439
AP: RCOut: PWM:1-12
AP: Frame: QUAD

In addition, when I change mode from Ground Computer, it works but it show nothing after typing the syntax on Xavier.
When I change mode from Ground Computer, the new mode still be synced on Xavier console.
Anyone has a clue?
Thank a lot,

garrettadamsss, 2022-11-27:
I have the same issue. Did you find a solution?