dronekit/dronekit-python #1053: is it possible to interrupt planned navigation , send mavlink messages with a path to follow or an obstacle, and then return control to planned navigation? Also how to get path data and send data.

Mytre, 2020-08-26:
I have some small questions. I have a lidar with 360 degrees, i want to send obstacle information using dronekit to a pixhawk using mavlink. I also want to get the path data , is this possible?  

Also, is there a way to interrupt a planned navigation , and guide the drone using code, and return then to planned navigation where it was left?

reubenfinch, 2020-09-02:
Are you wanting to return to the mission at the same transect (i.e. between the waypoints at point of changing to GUIDED?) If so you will need to keep some record of the next waypoint before changing mode, then advance to that waypoint again when switching back to AUTO.

Mytre, 2020-09-08:
> 
> 
> Are you wanting to return to the mission at the same transect (i.e. between the waypoints at point of changing to GUIDED?) If so you will need to keep some record of the next waypoint before changing mode, then advance to that waypoint again when switching back to AUTO.

That is my idea , I want to go into GUIDED mode, do some maneuvering then go back to AUTO mode.  How would I go about obtaining the waypoint or the list of waypoints?

lachlanbrewster1, 2020-11-25:
The way I do this is by having my onboard computer detect a servo pulse (using SERVO_OUTPUT_RAW) e.g. servo8 going from low to high, and then changing the mode in my code. So in the planned mission using mission planner: 
waypoint1, 
do_set_servo(8, high) 
delay(1)
do_set_servo(8, low)
waypoint2,
...