dronekit/dronekit-python #1058: Dronekit-Sitl " Error Disarming Motors "

drnayeem, 2020-09-07:
Hello

I am trying to connect to SITL.
The connect string works fine.
But when i try to send 
vehicle.armed = True 
command to arm the quad 

Error: autopilot : ARMING MOTORS
CRITICAL: autopilot : GROUND  START
CRITICAL: autopilot : Initializing APM...
ERROR: autopilot : DISARMING MOTORS

this is what i get 
am Running python 2.7 on a Windows10 PC

Charbel10, 2020-09-08:
Hello,

I'm trying to run the drone code "Simple_go_to" on python, but Im getting the following error and the alitude stays 0:

Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
Basic pre-arm checks
Arming motors
 Waiting for arming...
ERROR:autopilot:ARMING MOTORS
CRITICAL:autopilot:Initialising APM...
Taking off!
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
ERROR:autopilot:DISARMING MOTORS

Any help please?

Thnak you

leonardocfor, 2020-09-08:
Hi, can you guys provide your source code ?

Charbel10, 2020-09-08:
Hello,

Below is the code:

from __future__ import print_function
import time
from dronekit import connect, VehicleMode, LocationGlobalRelative

# Set up option parsing to get connection string
import argparse
parser = argparse.ArgumentParser(description='Commands vehicle using vehicle.simple_goto.')
parser.add_argument('--connect',
                    help="Vehicle connection target string. If not specified, SITL automatically started and used.")
args = parser.parse_args()

connection_string = args.connect
sitl = None

# Start SITL if no connection string specified
if not connection_string:
    import dronekit_sitl
    sitl = dronekit_sitl.start_default()
    connection_string = sitl.connection_string()

# Connect to the Vehicle
print('Connecting to vehicle on: %s' % connection_string)
vehicle = connect(connection_string, wait_ready=True)

def arm_and_takeoff(aTargetAltitude):
    """
    Arms vehicle and fly to aTargetAltitude.
    """

    print("Basic pre-arm checks")
    # Don't try to arm until autopilot is ready
    while not vehicle.is_armable:
        print(" Waiting for vehicle to initialise...")
        time.sleep(1)

    print("Arming motors")
    # Copter should arm in GUIDED mode
    vehicle.mode = VehicleMode("GUIDED")
    vehicle.armed = True

    # Confirm vehicle armed before attempting to take off
    while not vehicle.armed:
        print(" Waiting for arming...")
        time.sleep(1)

    print("Taking off!")
    vehicle.simple_takeoff(aTargetAltitude)  # Take off to target altitude

    # Wait until the vehicle reaches a safe height before processing the goto
    #  (otherwise the command after Vehicle.simple_takeoff will execute
    #   immediately).
    while True:
        print(" Altitude: ", vehicle.location.global_relative_frame.alt)
        # Break and return from function just below target altitude.
        if vehicle.location.global_relative_frame.alt >= aTargetAltitude * 0.95:
            print("Reached target altitude")
            break
        time.sleep(1)

arm_and_takeoff(10)

print("Set default/target airspeed to 3")
vehicle.airspeed = 3

print("Going towards first point for 30 seconds ...")
point1 = LocationGlobalRelative(-35.361354, 149.165218, 20)
vehicle.simple_goto(point1)

# sleep so we can see the change in map
time.sleep(30)

print("Going towards second point for 30 seconds (groundspeed set to 10 m/s) ...")
point2 = LocationGlobalRelative(-35.363244, 149.168801, 20)
vehicle.simple_goto(point2, groundspeed=10)

# sleep so we can see the change in map
time.sleep(30)

print("Returning to Launch")
vehicle.mode = VehicleMode("RTL")

# Close vehicle object before exiting script
print("Close vehicle object")
vehicle.close()

# Shut down simulator if it was started.
if sitl:
    sitl.stop()

leonardocfor, 2020-09-08:
Ok I assume you are using the simple_goto.py script in the examples folder. I personally don't like much dronekit-sitl, but I find dronekit to work quite good with ArduPilot SITL, which you can install it by following this link: 

https://ardupilot.org/dev/docs/building-setup-linux.html#building-setup-linux (Assuming you are using Linux). 

Then run your code with the --connect flag  using <YOUR_IP_OR_SIMPLY_localhost>:PORT, where port can be 14550 e.g.:

python simple_goto.py --connect localhost:14550

Charbel10, 2020-09-08:
Thk you for your quick reply.
Iam using windows and not linux. I'm sorry Im a beginner in python.
In addition, Im using the mission planner as simulator.
I really need to run properly this example.

Thank you for your support,

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif]<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>    Virus-free. www.avast.com<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>

leonardocfor, 2020-09-08:
You can try install ArduPilot SITL in Windows by following this link: https://ardupilot.org/dev/docs/sitl-native-on-windows.html 

Or you can use virtual box or docker to run an Ubuntu in your Windows and then install ArduPilot SITL on it and run the DroneKit examples. I would recommend you to test this software on Linux nevertheless as in reality it would be unusual to have a drone running Windows hehe.

drnayeem, 2020-09-09:
Hello 
Thanks A lot **leonardocfor**
I tried Installing Ardupilot Sitl following the steps 
1. Install Cigwin 
2. Install MAVProxy 
and then we are supposed to call sim_vehicle.py from Cigwin 
But i did not get any directory called ' Vehicles' and containing the Vehicle_sim.py file 

Am i doing something wrong ?

Regards 
Doc

CordobaPablo, 2020-11-11:
Did you find the solution?

TulipDi, 2020-11-20:
I get the same issue, have you solved it?

drnayeem, 2020-11-20:
No it's still stuck at that 
Awaiting some way out

asosa16, 2021-01-13:
Set your vehicle mode to STABILIZE (or any other armable mode).
Run your python script.
Immediately change your vehicle mode manually to GUIDED.
Your vehicle should perform your script commands if correct.

This solution is far from ideal, but I've found it to work. Good luck

GRS26, 2021-01-24:
@leonardocfor could you explain a little better how to setup this ardupilot sitl?

m-s-spoorthi, 2022-10-26:
after the "vehicle.simple_takeoff(aTargetAltitude)  # Take off to target altitude"
give enough time, at least 30s to take off
ex: time.sleep(30) after the command

prithvi2k2, 2022-12-08:
A potential fix could be downgrading pymavlink to 2.4.8
Try
`pip uninstall pymavlink`
then
`pip install pymavlink==2.4.8`

Usage of `pip` command depends on your environment setup defaults, I assume you're on python 2.7 as you mentioned
Another observation - If you're using **python 3**, the above workaround might not work with python 3.7+, will work with upto python 3.6

vijayakashallenki, 2022-12-13:
Thank you man its working now!!!

kyrlon, 2023-02-22:
Downgrading to python3.6 is not ideal. Is there a dronekit-sitl docker container available by chance?