dronekit/dronekit-python #1059: Python 3 compatibility?

Mytre, 2020-09-10:
Hello, I have a small question. Is this compatible with python 3? and if so, is there any documentation with python 3 in mind?

peterbarker, 2020-09-10:
Should be Python3 compatible.  Raise an issue if you find a problem.

closed 2020-09-10 by peterbarker

rspandolfi, 2021-01-04:
We have been sticking with Python 2 unsure of what problems we might experience with DroneKit should we upgrade to Python 3. If you are reasonably comfortable that DroneKit will work with Python 3, perhaps you could issue a migration guide as you did for Python 1 to Python 2? https://dronekit-python.readthedocs.io/en/latest/

Mytre, 2021-01-04:
Hello, sorry for the long response, yes it is compatible, a bit of issues mostly regarding documentation on how for example to use mavlink 2 messages such as the more complex obstacle message.  It worked as expected in any case only using the simple obstacle message on mavlink 1 that has the sector based detection.