dronekit/dronekit-python #1065: Connecting to a Vehicle in a Multi Vehicle Link

mustafa-gokce, 2020-09-21:
Hi.
I am using RFD900x telemetry modules to connect multiple vehicles one by one.
I am using the the following commands.
To connect to telemetry:
mavproxy.exe --master=COM256,57600 --out=127.0.0.1:10000 --out=127.0.0.1:10010 --out=127.0.0.1:10020 --out=127.0.0.1:10030 --cmd="set shownoise false" --daemon
To connect and distribute vehicle one by one:
mavproxy.exe --master=udpin:127.0.0.1:10010 --target-system=10 --cmd="set shownoise false" --out=127.0.0.1:20010
mavproxy.exe --master=udpin:127.0.0.1:10020 --target-system=20 --cmd="set shownoise false" --out=127.0.0.1:20020
mavproxy.exe --master=udpin:127.0.0.1:10030 --target-system=30 --cmd="set shownoise false" --out=127.0.0.1:20030
I assumed that above commands only outputs the target systems one by one but it is not.
When i try to connect the vehicles using below lines of codes:
vehicle1 = connect('127.0.0.1:20010', wait_ready=True)
vehicle2 = connect('127.0.0.1:20020', wait_ready=True)
vehicle3 = connect('127.0.0.1:20030', wait_ready=True)
The incoming and outgoing data traffic is mixing and there is no target_system parameter for connect() function.
How can i connect vehicles one by one without mixing them up?
Thanks...

eminakgun, 2020-11-12:
What dronekit does is,
It assumes there is a single vehicle in that connection and waits for its heartbeat to [detect ](https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L2274)its system_id. Once captured the system, it uses that system_id when sending messages that requires target_system field. For incoming and outgoing messages that does not include target fields, it thinks that its the same vehicle. So that's why your traffic is mixing.

What I suggest,
If you don't know system ids in that connection, first you need to search which id's are connected by looking at the HEARTBEAT messages in the connection and you can then use those ids to spawn separate vehicle classes with the implementation that does accept the [target_system](https://github.com/dronekit/dronekit-python/blob/27b292b8eb32e465a3682e84bab3a56fb83bba0c/dronekit/mavlink.py#L124) in Class MAVConnection and you have to filter out the ids in [input ](https://github.com/dronekit/dronekit-python/blob/27b292b8eb32e465a3682e84bab3a56fb83bba0c/dronekit/mavlink.py#L205)and [output ](https://github.com/dronekit/dronekit-python/blob/27b292b8eb32e465a3682e84bab3a56fb83bba0c/dronekit/mavlink.py#L170)loops.

GRS26, 2022-06-01:
@eminakgun Hi, I have some questions about this topic, and don't want to cluster the discussion with them, if it's not a bother and you are available and willing, could you provide some way of contacting?

eminakgun, 2022-06-04:
I don't think there is a much of serious discussion here and its reeeealy been long time since I commented. So drop your questions here and I would answer if I can do so. But as I said, I am not into this topic for a reaaly long time.

GRS26, 2022-06-05:
Hello,
First of all, much thanks for taking your time to indulge my questions. 

For context, my setup is made of a companion computer connected to a flight controller,using dronekit, transmitting through a radio link to another end which extracts the message through a dronekit script. What I want to achieve is send a mavlink message, battery_status, from companion computer to flight controller, then extract that specific message.

What is happening is : at the other end, the message is being extracted, but not the one originating from my companion computer, but the one from the flight controller. It seems the problem is that dronekit automatically targets the system id 0, the flight controller, and does not allow to change it. Using Mission Planner I can see that message is within the stream, but coming from vehicle 255.

Basically I want to make this work, it doesn't even need to solve this multiple vehicle thing, I just want to be able to send data while taking advantage of the mavlink stream because my setup does not allow for other radio link just to send this message.

I tried to understand your instructions, but couldn't quite figure it out, I spent some days learning about threading, queue and reading the pymavlink and mavuitl code, but it seems my skills are still beneath what is required to solve this.

So , within your capacity, do you have any advice? I thought about making a custom mavlink message, but I am afraid I will run into the same multiple vehicle problem, while I am not sure if this would really happen. Do you think a custom message could work?

eminakgun, 2022-06-06:
Applications in a multi-vehicle environment definitely requires extra coding work. Dronekit is not designed for it.
Adding a custom messages are straight forward, though. But I don't think its necessary for your application. In a multi-vehicle environment, you have to filter out the coming messages in your dronekit application(As I have mentioned in previous comment). Because, AFAIK, AP considers messages with no target fields as broadcast message and routes it to your telemetry and to CC (if message is received from radio). So check out your autopilot's telemetry message routing policy. 

And, Dronekit's default implementation constrtucts MAVConnection or some class in pymavlink with system_id set to 255. I think that's why you see that number in GCS, you send a message from dronekit to your AP and it routes it to telemetry then it shows up in your GCS.

If you add a custom message, let's say having target system and component fields, you will end with a situation like,
1. Send that custom message to AP and AP won't accept it since target fields does not match, and route it to its telemetry point
2. Other end AP will receive and won't accept then route it to CC, so your CC(input loop) will receive it.
3. In [input loop](https://github.com/dronekit/dronekit-python/blob/27b292b8eb32e465a3682e84bab3a56fb83bba0c/dronekit/mavlink.py#L205), you have to figure out the message is not coming from the AP that you are attached to.
4. This requires your application to know the system_id of your AP you are connected to. In order to do that you need to power-up the the AP-CC pairs respectively so that they recognize correct system_id's, not other AP-CC pair, since there are broadcast messages. Or, from scratch, you gotta know and specify the correct system_id when connecting.
5. Once you detect that message is coming from some remote CC, then you need to figure out how to send that message object to your particular application code. (I don't remember how to!)

Without a custom message, you only need the message going out of your CC to have different system_id and component_it rather than 255, so that you can differentiate ids in remote application, the same way as above. You can accomplish it when connecting to AP from dronekit. You may need to dive in to the source code, I don't remember where it resides.

Hope that helps. Multi-vehicle application is not straight-forward. If you start a work about this topic, ask a code review in your repository from me as you commit. I may guide you.