dronekit/dronekit-python #1066: ERROR:autopilot:ARMING MOTORS

atomattias, 2020-09-23:
Help me out, I am new to Dronekit-python, My goal is to use dronekit python on sitl. 
My system is ubuntu 18.04
I am running  _$dronekit-sitl copter_ in one terminal and tried to connect with it from another. 
I test it with the sample script.
_$python simple_goto.py_
But it looks like it is not changing mode and is failed as it tried to arm the motor.

[image]

sakaryag, 2020-09-28:
Did you solve this problem? I am taking this error too.

loicspace, 2020-10-05:
Same issue for me, for this script specifically and my own, which all run successfully on an older install.

xbalaguer, 2020-11-06:
Same error... when trying to takeoff the SITL, is not gaining altitude due to the same problem. It was working before the last update.

DavidM-GitHub, 2020-11-30:
For me this was caused by the vehicle state not changing to guided mode. I reinstalled an older version of mavlink as per this comment https://github.com/dronekit/dronekit-python/issues/1041#issuecomment-660750737. This solved the problem for me

danammeansbear, 2021-01-15:
This problem has persisted but when I downgrade, it produces a haystack of a mess

habibmuhammadthariq, 2021-03-24:
> For me this was caused by the vehicle state not changing to guided mode. I reinstalled an older version of mavlink as per this comment [#1041 (comment)](https://github.com/dronekit/dronekit-python/issues/1041#issuecomment-660750737). This solved the problem for me

thanks @DavidM-GitHub , this work for me

swere10, 2023-12-14:
Hello
is there anyone with a solution to this problem