dronekit/dronekit-python #1069: error on trying to run drone await set(types) invalid syntax error

prashanthravi19, 2020-09-28:
i get an error while trying to run drone delivery.py how can the bug be resolved?
 await = set(types)
          ^
SyntaxError: invalid syntax
full stack trace
Traceback (most recent call last):
  File "drone_delivery.py", line 18, in <module>
    from dronekit import connect, VehicleMode, LocationGlobal, LocationGlobalRelative
  File "C:\Users\Prashanth\AppData\Local\Programs\Python\Python38\lib\site-packages\dronekit\__init__.py", line 2192
    await = set(types)
          ^
SyntaxError: invalid syntax

4NTH3L1OS, 2020-09-28:
I think you should use Python 2.7 to run your code. Can you try it? There is a some syntax differences between Python 3.x and Python 2.x. The error may be caused by this situation.

prashanthravi19, 2020-09-29:
thank u i ve made changes the app is running but when i run dronekit delivery it gets stuck on polled  mode guided

jinkalr-itpath, 2020-10-01:
I want to connect DJI drone, can I use droneKit API? its urgent please